mavlink/qgroundcontrol #8772: ADS-B airplane traces

theDimitron, 2020-05-25:
Hi,
In my project I'm using FLARM device to track all airplanes above me in the sky. I publish the vehicles data thought transponder_report topic and airplane image appears on the map. The vehicle leaves some kind of “traces” of its images on a map, which disappear in 30-45 sec. I’m using QGC v4.0.6. Does anyone know why it happens?
The interesting part for me, when I build same version [4.0.6] of QCG with Qt the problem is no longer there. 
- Operating System: Windows 10
- QGC Version: 4.0.6 and I’ve also built from master 
- Flight Controller: Pixhawk 4
- Autopilot: PX4 1.10.2dev
[image]

DonLakeFlyer, 2020-05-25:
Attach a telemetry log which has the ADSB telemetry in it and I can take a look.

theDimitron, 2020-05-26:
Can you please tell me how to do log for ADS-B messages? I can simply start a logging in QGC but I cannot find any useful information on ADS-B in it.

DonLakeFlyer, 2020-05-26:
I need a telemetry log not a console log: https://docs.qgroundcontrol.com/en/SettingsView/General.html. Turn on  "save logs even if vehicle not armed". Run QGC with ADSB telemetry coming in. Close QGC. Attach/link the telemetry log here.

theDimitron, 2020-05-26:
https://review.px4.io/plot_app?log=5c71628b-e44e-4641-8959-239723a6932c

DonLakeFlyer, 2020-05-26:
This is a vehicle binary ulog not a QGC mavlink telemetry log.

DonLakeFlyer, 2020-05-26:
Telemetry logs will be saved in your QGroundControl/Telemetry folder in your Documents folder which is in different places depending on OS.

theDimitron, 2020-05-26:
Thank you for tip:) 

Here is the file:
[2020-05-26 21-35-15.zip](https://github.com/mavlink/qgroundcontrol/files/4685174/2020-05-26.21-35-15.zip)

DonLakeFlyer, 2020-05-26:
QGC Bug. Working on a fix.

DonLakeFlyer, 2020-05-29:
The fix is available in latest Stable. Daily will update some time today.

smagellan, 2021-08-21:
@DonLakeFlyer, by the way: there are no callsigns on the initial screenshot; v4.13 has no callsigns, too. In the meantime, v3.3 had this feature: https://docs.qgroundcontrol.com/master/en/releases/stable_v3.3_long.html#ads-b-sensor-vehicle-display. Was it removed intentionally?

smagellan, 2021-09-17:
@DonLakeFlyer, I may take a look at this, just let me know if it is ok for you

closed 2021-09-21 by DonLakeFlyer

smagellan, 2021-10-02:
@DonLakeFlyer, done: https://github.com/mavlink/qgroundcontrol/pull/9897