mavlink/qgroundcontrol #8778: Anroid build error

akontsevich, 2020-05-25:
Android version builds and links fine, however fails on deployment step - building android package with following error:

```
Building the android package failed!
  -- For more information, run this command with --verbose.
  -- Skipping /home/aleksey/Qt/5.12.8/android_arm64_v8a/plugins/position/libqtposition_serialnmea.so. It has unmet dependencies: lib/libQt5SerialPort.so.
```

DonLakeFlyer, 2020-05-25:
This seems to be working fine in CI. So it may have something to do with using 5.12.8 versus 5.12.6 which is the current supported version of Qt.

akontsevich, 2020-05-25:
> This seems to be working fine in CI. So it may have something to do with using 5.12.8 versus 5.12.6 which is the current supported version of Qt.

No, same bug there:

```
Error: Could not find or load main class org.gradle.wrapper.GradleWrapperMain
Building the android package failed!
  -- For more information, run this command with --verbose.
  -- Skipping /home/aleksey/Qt/5.12.6/android_arm64_v8a/plugins/position/libqtposition_serialnmea.so. It has unmet dependencies: lib/libQt5SerialPort.so.
```

Any other ideas?

akontsevich, 2020-05-25:
Success! Solved, could be wrong symbol in build dir name. Sorry.

closed 2020-05-25 by akontsevich