mavlink/qgroundcontrol #8789: Mac 10.15.4: QGroundControl quit unexpectedly

srosro, 2020-05-28:
* Pixhawk 2.1 Orange Cube
* Connected via USB-C

Crash report attached.

[image]

```Process:               QGroundControl [751]
Path:                  /Users/USER/Desktop/QGroundControl.app/Contents/MacOS/QGroundControl
Identifier:            org.qgroundcontrol.qgroundcontrol
Version:               4.0.6 (???)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           QGroundControl [751]
User ID:               501

Date/Time:             2020-05-27 17:44:38.511 -0700
OS Version:            Mac OS X 10.15.4 (19E287)
Report Version:        12
Bridge OS Version:     3.0 (14Y908)
Anonymous UUID:        1BB6DA35-5B6D-7D5C-BB04-3DD52EFAE597

Time Awake Since Boot: 350 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000340000000112
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [751]

VM Regions Near 0x340000000112:
    JS VM Gigacage         0000000121e67000-0000000122267000 [ 4096K] rw-/rwx SM=PRV  
--> 
    MALLOC_NANO            0000600000000000-0000600008000000 [128.0M] rw-/rwx SM=PRV  

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   org.qt-project.QtQml          	0x00000001125a98f4 QQmlData::isSignalConnected(QAbstractDeclarativeData*, QObject const*, int) + 4
1   org.qt-project.QtCore         	0x0000000112c67cea QMetaObject::activate(QObject*, int, int, void**) + 90
2   org.qgroundcontrol.qgroundcontrol	0x000000010f407ecf 0x10f3d8000 + 196303
3   org.qgroundcontrol.qgroundcontrol	0x000000010f5553ee 0x10f3d8000 + 1561582
4   org.qgroundcontrol.qgroundcontrol	0x000000010f55cd1f 0x10f3d8000 + 1592607
5   org.qt-project.QtCore         	0x0000000112c6853b QMetaObject::activate(QObject*, int, int, void**) + 2219
6   org.qgroundcontrol.qgroundcontrol	0x000000010f58cce2 0x10f3d8000 + 1789154
7   org.qgroundcontrol.qgroundcontrol	0x000000010f584d12 0x10f3d8000 + 1756434
8   org.qt-project.QtCore         	0x0000000112c6853b QMetaObject::activate(QObject*, int, int, void**) + 2219
9   org.qgroundcontrol.qgroundcontrol	0x000000010f5b3446 0x10f3d8000 + 1946694
10  org.qt-project.QtCore         	0x0000000112c6853b QMetaObject::activate(QObject*, int, int, void**) + 2219
11  org.qt-project.QtSerialPort   	0x00000001129e818f 0x1129e0000 + 33167
12  org.qt-project.QtSerialPort   	0x00000001129ea726 0x1129e0000 + 42790
13  org.qt-project.QtWidgets      	0x000000011178d77d QApplicationPrivate::notify_helper(QObject*, QEvent*) + 269
14  org.qt-project.QtWidgets      	0x000000011178eb82 QApplication::notify(QObject*, QEvent*) + 594
15  org.qt-project.QtCore         	0x0000000112c37574 QCoreApplication::notifyInternal2(QObject*, QEvent*) + 212
16  org.qt-project.QtCore         	0x0000000112c8a507 0x112a4d000 + 2348295
17  com.apple.CoreFoundation      	0x00007fff33a4fc75 __CFSocketPerformV0 + 942
18  com.apple.CoreFoundation      	0x00007fff33a24f12 __CFRUNLOOP_IS_CALLING_OUT_TO_A_SOURCE0_PERFORM_FUNCTION__ + 17
19  com.apple.CoreFoundation      	0x00007fff33a24eb1 __CFRunLoopDoSource0 + 103
20  com.apple.CoreFoundation      	0x00007fff33a24ccb __CFRunLoopDoSources0 + 209
21  com.apple.CoreFoundation      	0x00007fff33a239fa __CFRunLoopRun + 927
22  com.apple.CoreFoundation      	0x00007fff33a22ffe CFRunLoopRunSpecific + 462
23  com.apple.HIToolbox           	0x00007fff32656abd RunCurrentEventLoopInMode + 292
24  com.apple.HIToolbox           	0x00007fff326567d5 ReceiveNextEventCommon + 584
25  com.apple.HIToolbox           	0x00007fff32656579 _BlockUntilNextEventMatchingListInModeWithFilter + 64
26  com.apple.AppKit              	0x00007fff30ca1c99 _DPSNextEvent + 883
27  com.apple.AppKit              	0x00007fff30ca04e0 -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 1352
[... 607 lines cut ...]
TOTAL                              1.5G     1746 
TOTAL, minus reserved VM space     1.3G     1746 

Model: MacBookPro14,2, BootROM 205.0.0.0.0, 2 processors, Dual-Core Intel Core i7, 3.5 GHz, 16 GB, SMC 2.44f4
Graphics: kHW_IntelIrisGraphics650Item, Intel Iris Plus Graphics 650, spdisplays_builtin
Memory Module: BANK 0/DIMM0, 8 GB, LPDDR3, 2133 MHz, 0x802C, 0x4D5435324C31473332443450472D30393320
Memory Module: BANK 1/DIMM0, 8 GB, LPDDR3, 2133 MHz, 0x802C, 0x4D5435324C31473332443450472D30393320
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x171), Broadcom BCM43xx 1.0 (7.77.111.1 AirPortDriverBrcmNIC-1600.4)
Bluetooth: Version 7.0.4f6, 3 services, 27 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
USB Device: USB3.1 Hub
USB Device: USB 10/100/1000 LAN
USB Device: USB 3.0 Bus
USB Device: USB2.1 Hub
USB Device: Hub
USB Device: Kinesis Keyboard Hub
USB Device: Composite Device
USB Device: Apple LED Cinema Display
USB Device: Display iSight
USB Device: Display Audio
USB Device: CubeOrange
USB Device: Apple T1 Controller
Thunderbolt Bus: MacBook Pro, Apple Inc., 41.4
Thunderbolt Bus: MacBook Pro, Apple Inc., 41.4
```

srosro, 2020-05-28:
I was actually able to get the crash on video: https://www.dropbox.com/s/d2fwyzoxwbkx5o8/Error%20report%20-%20QGroundControl.mov?dl=0

A few interesting things in the video:
* `Your Vehicle is not responding` appears, even though QGroundControl seems to show uninterrupted gyroscope/compass readings.
* That error message was appearing since day 1 of QGroundControl install, but does not seem to otherwise affect functionality.  In attempt to fix, I installed latest 64-bit FTDI drivers here: https://www.ftdichip.com/Drivers/VCP.htm. After install of 64-bit drivers, QGroundControl is now crashing (as seen in video).
* Yes, I love my wife.

srosro, 2020-05-28:
Okay, another update - so I've deleted the driver I installed: `sudo mv /Library/Extensions/FTDIUSBSerialDriver.kext ~/Desktop/`

QGroundControl no longer crashes, but I still get the weird original behavior, which includes several error messages:
* `Vehicle failed to send AUTOPILOT_VERSION after waiting/retrying for 10 seconds`
* `Your Vehicle is not responding.  If this continues, shutdown QGroundControl, restart the Vehicle letting it boot completely, then start QGroundControl.`

See this video: https://www.dropbox.com/s/6ooyx2b7c3caj9z/Still%20get%20%22failed%20to%20connect%22.mov?dl=0

Following the instructions (shutdown, restart, etc) does not fix the issue as the error message returns (can record this as well if needed).

srosro, 2020-05-28:
Flashing the Pixhawk 2.1 Cube with the ArduPilot beta / developer builds do not fix the issue and instead introduce new error messages.  See screenshot:
 
[image]

srosro, 2020-05-28:
Edit: spoke too soon.  Removing the driver might improve the crash rate (as QGroundControl used to crash after just a few minutes before removing driver), but does not completely fix things.

QGroundControl just crashed after ~4100 seconds.  Latest crash report:

```
Process:               QGroundControl [803]
Path:                  /Users/USER/Desktop/QGroundControl.app/Contents/MacOS/QGroundControl
Identifier:            org.qgroundcontrol.qgroundcontrol
Version:               4.0.6 (???)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           QGroundControl [803]
User ID:               501

Date/Time:             2020-05-27 19:16:11.653 -0700
OS Version:            Mac OS X 10.15.4 (19E287)
Report Version:        12
Bridge OS Version:     3.0 (14Y908)
Anonymous UUID:        1BB6DA35-5B6D-7D5C-BB04-3DD52EFAE597

Time Awake Since Boot: 4100 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x00004792c8088120
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [803]

VM Regions Near 0x4792c8088120:
    MALLOC_LARGE           000000011951f000-000000011970e000 [ 1980K] rw-/rwx SM=PRV  
--> 
    MALLOC_NANO            0000600000000000-0000600008000000 [128.0M] rw-/rwx SM=PRV  

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   org.qt-project.QtCore         	0x0000000109649ca8 QMetaObject::cast(QObject*) const + 24
1   org.qgroundcontrol.qgroundcontrol	0x0000000105f857a1 0x105e44000 + 1316769
2   org.qgroundcontrol.qgroundcontrol	0x00000001060e4051 0x105e44000 + 2752593
3   org.qt-project.QtCore         	0x000000010967453b QMetaObject::activate(QObject*, int, int, void**) + 2219
4   org.qt-project.QtCore         	0x000000010967c432 QTimer::timerEvent(QTimerEvent*) + 98
5   org.qt-project.QtCore         	0x000000010966ccf6 QObject::event(QEvent*) + 102
6   org.qt-project.QtWidgets      	0x00000001081cf77d QApplicationPrivate::notify_helper(QObject*, QEvent*) + 269
7   org.qt-project.QtWidgets      	0x00000001081d0b82 QApplication::notify(QObject*, QEvent*) + 594
8   org.qt-project.QtCore         	0x0000000109643574 QCoreApplication::notifyInternal2(QObject*, QEvent*) + 212
9   org.qt-project.QtCore         	0x000000010969cb7f QTimerInfoList::activateTimers() + 991
10  libqcocoa.dylib               	0x000000010c188262 0x10c154000 + 213602
11  com.apple.CoreFoundation      	0x00007fff36bccf12 __CFRUNLOOP_IS_CALLING_OUT_TO_A_SOURCE0_PERFORM_FUNCTION__ + 17
12  com.apple.CoreFoundation      	0x00007fff36bcceb1 __CFRunLoopDoSource0 + 103
13  com.apple.CoreFoundation      	0x00007fff36bccccb __CFRunLoopDoSources0 + 209
14  com.apple.CoreFoundation      	0x00007fff36bcb9fa __CFRunLoopRun + 927
15  com.apple.CoreFoundation      	0x00007fff36bcaffe CFRunLoopRunSpecific + 462
16  com.apple.HIToolbox           	0x00007fff357feabd RunCurrentEventLoopInMode + 292
17  com.apple.HIToolbox           	0x00007fff357fe7d5 ReceiveNextEventCommon + 584
18  com.apple.HIToolbox           	0x00007fff357fe579 _BlockUntilNextEventMatchingListInModeWithFilter + 64
19  com.apple.AppKit              	0x00007fff33e49c99 _DPSNextEvent + 883
20  com.apple.AppKit              	0x00007fff33e484e0 -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 1352
21  com.apple.AppKit              	0x00007fff33e3a1ee -[NSApplication run] + 658
22  libqcocoa.dylib               	0x000000010c189013 0x10c154000 + 217107
23  org.qt-project.QtCore         	0x000000010963eb8f QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) + 431
24  org.qt-project.QtCore         	0x0000000109643b82 QCoreApplication::exec() + 130
25  org.qgroundcontrol.qgroundcontrol	0x0000000106000e86 0x105e44000 + 1822342
26  org.qgroundcontrol.qgroundcontrol	0x0000000105e4d374 start + 52

Thread 1:: AMCP Logging Spool
0   libsystem_kernel.dylib        	0x00007fff70abbe36 semaphore_wait_trap + 10
[... 590 lines cut ...]
TOTAL                              1.5G     1958 
TOTAL, minus reserved VM space     1.4G     1958 

Model: MacBookPro14,2, BootROM 205.0.0.0.0, 2 processors, Dual-Core Intel Core i7, 3.5 GHz, 16 GB, SMC 2.44f4
Graphics: kHW_IntelIrisGraphics650Item, Intel Iris Plus Graphics 650, spdisplays_builtin
Memory Module: BANK 0/DIMM0, 8 GB, LPDDR3, 2133 MHz, 0x802C, 0x4D5435324C31473332443450472D30393320
Memory Module: BANK 1/DIMM0, 8 GB, LPDDR3, 2133 MHz, 0x802C, 0x4D5435324C31473332443450472D30393320
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x171), Broadcom BCM43xx 1.0 (7.77.111.1 AirPortDriverBrcmNIC-1600.4)
Bluetooth: Version 7.0.4f6, 3 services, 27 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
USB Device: USB3.1 Hub
USB Device: USB 10/100/1000 LAN
USB Device: USB 3.0 Bus
USB Device: USB2.1 Hub
USB Device: Hub
USB Device: Kinesis Keyboard Hub
USB Device: Composite Device
USB Device: Apple LED Cinema Display
USB Device: Display iSight
USB Device: Display Audio
USB Device: CubeOrange
USB Device: Apple T1 Controller
Thunderbolt Bus: MacBook Pro, Apple Inc., 41.4
Thunderbolt Bus: MacBook Pro, Apple Inc., 41.4
```

DonLakeFlyer, 2020-05-28:
It seems like QGC may be connecting to something else on your computer thinking it is a Vehicle. Can you turn on LinkManagerLog and attach the console log output: https://docs.qgroundcontrol.com/en/SettingsView/console_logging.html

DonLakeFlyer, 2020-05-28:
COMPASS_PRIMARY error seems to be something new with latest master ArduPilot. They may have removed a parameter that QGC needs. Happens all the time with new versions unfortunately. Haven't looked at it yet to figure it out.

srosro, 2020-05-28:
Log attached.  I did get the error message after the loging was enabled, and before I saved the log, so hopefully the log provides some insight:

[image]

[qgroundcontrol_log.txt](https://github.com/mavlink/qgroundcontrol/files/4694050/qgroundcontrol_log.txt)

DonLakeFlyer, 2020-05-30:
You see here from the log
```
[D] at /Users/travis/build/mavlink/qgroundcontrol/src/comm/LinkManager.cc:572 - "Waiting for next autoconnect pass "/dev/cu.usbmodem1454201""
[D] at /Users/travis/build/mavlink/qgroundcontrol/src/comm/LinkManager.cc:572 - "Waiting for next autoconnect pass "/dev/cu.usbmodem1454203""
[D] at /Users/travis/build/mavlink/qgroundcontrol/src/comm/LinkManager.cc:613 - "New auto-connect port added:  "Hex/ProfiCNC on cu.usbmodem1454201 (AutoConnect)" "/dev/cu.usbmodem1454201""
[D] at /Users/travis/build/mavlink/qgroundcontrol/src/comm/LinkManager.cc:613 - "New auto-connect port added:  "Hex/ProfiCNC on cu.usbmodem1454203 (AutoConnect)" "/dev/cu.usbmodem1454203""
```

That QGC is connect is somehow connecting to what it thinks are two cubes.

My thinking is that you machine is hosed up somehow by installing that FTDI driver still. You can try taking QGC out of the equation. Using Terminal:
* Before connecting Cube
* Do `ls /dev` and see what is listed
* Connect the Cube via USB
* Do `ls /dev` again and see if you see two usb ports added.

If you do, then you have some strange OS problem.

srosro, 2020-06-03:
Okay, this is what I have right now.  Let me dig in further:

```
$ ls /dev | grep usb
(no results - then plug in cube)
$ ls /dev | grep usb
cu.usbmodem146201
cu.usbmodem146203
tty.usbmodem146201
tty.usbmodem146203
```

srosro, 2020-06-04:
Well I've confirmed the FTDI driver has been removed (removing the kext driver file and using pkgutil --forget):

`$ pkgutil --pkgs | grep -i ftdi`

Returns nothing.  Great.  Then I thought I'd poke around network interfaces and found this:

[image]
[image]

I removed all of those networks, but I can't seem to figure out a way to get rid of the interface driver still:

[image]

Curious if anyone has any tips.  I can't figure out why the Cube still comes up when its driver was uninstalled.  I also can't find any other usb driver that seems cube related:

```
$ ls /Library/Extensions/ | grep usb -i
BJUSBLoad.kext
CIJUSBLoad.kext
$ ls /System/Library/Extensions/ | grep usb -i
AppleMIDIUSBDriver.plugin
AppleOSXUSBNCM.kext
AppleUSBACM.kext
AppleUSBAudio.kext
AppleUSBCDC.kext
AppleUSBCommon.kext
AppleUSBDMM.kext
AppleUSBDisplays.kext
AppleUSBECM.kext
AppleUSBEEM.kext
AppleUSBEthernet.kext
AppleUSBEthernetHost.kext
AppleUSBMultitouch.kext
AppleUSBNCM.kext
AppleUSBNetworking.kext
AppleUSBPLCOM.kext
AppleUSBRealtek8153Patcher.kext
AppleUSBSerial.kext
AppleUSBTopCase.kext
AppleUSBWCM.kext
AppleUSBiBridge.kext
IOUSBAttachedSCSI.kext
IOUSBFamily.kext
IOUSBHostFamily.kext
IOUSBMassStorageClass.kext
IOUSBMassStorageDriver.kext
```

Would love any tips on how to get this driver uninstalled...

DonLakeFlyer, 2020-06-08:
> I would love any tips on how to get this driver uninstalled so that I can then reinstall it...

There is no need to install any additional drivers on OSX to get boards to connect. It just works.

From the last screenshot os OSX network add that would seem odd the Cube shows up in that list. I have every Cube color other than Orange and they don't show up there.

All of that said you should be able to turn auto-connect and then create a manual connection to your device. That may work around the crash.

DonLakeFlyer, 2020-06-09:
FYI: I've been talking with the Cube folks. There may be some difference with they Orange cube. They are sending me one to test with so hopefully I can reproduce what you are seeing.

DonLakeFlyer, 2020-06-09:
So the two ports showing up is correct. One is for mavlink and the other is for SLCAN. This means if you figure out which is Mavlink turning autoconnect off and manually connect to the mavlink device should work. Until I get my Orange in the mail and can figure out how to deal with it.

DonLakeFlyer, 2020-06-09:
* Can you turn on SerialLinkLog
* Boot QGC with usb not connect
* Connect to Orange over USB
* Save/attach log here

That might give me enough info to try some things before I get my Orange.

DonLakeFlyer, 2020-06-13:
Ok, ProfiCNC send me an Orange Cube. The Vehicle not responding is from the SLCAN port which QGC things is a vehicle. I'll figure that out., It's not crashing for me though. Did the crash happen fairly quickly after connect? 

Also what is the exact sequence:
* Plug in board to USB, boot QGC?
or
* Boot QGC, plug in board to USB

srosro, 2020-06-14:
I'm wondering if the crash is a separate issue.  QGroundControl just crashed for me without the Orange Cube plugged in.  I'm currently using the knockoff 2.4.8 Pixhawk.  Steps prior:

* Edited a parameter (GPS_TYPE)
* Rebooted vehicle (Parameters > Tools > Reboot vehicle)
* QGroundControl crashed.

Crash report attached.
[crash.txt](https://github.com/mavlink/qgroundcontrol/files/4776458/crash.txt)

DonLakeFlyer, 2020-06-15:
Hmm, that looks the same as the other crash. Let me see if I can get you a debug build to install so I can get better stack trace symbols.

DonLakeFlyer, 2020-06-15:
Can you install this someplace and run it to try to get a crash stack trace with it: https://qgroundcontrol.s3-us-west-2.amazonaws.com/DonLakeFlyer/QGroundControl.dmg. This is a Stable debug build with symbols.

srosro, 2020-06-15:
> trace with it: https:/ / qgroundcontrol. s3-us-west-2. amazonaws. com/ DonLakeFlyer/
> QGroundControl. dmg (

Will do.

srosro, 2020-06-15:
Okay, installed from the above URL.  Just crashed.  Report attached.  Let me know if you need anything else.

[crash.txt](https://github.com/mavlink/qgroundcontrol/files/4783242/crash.txt)

DonLakeFlyer, 2020-06-15:
Crap I didn't get symbols. I'm going to have to play around with some more to figure out how to get you a package with symbols in it.

DonLakeFlyer, 2020-06-19:
Ok, there is a new dmg at the same link. Download that one, open and you have a new .app and a .dSYM. Copy both of those out of there to the same place. Run QGC, crash and hopefully the stack trace will now have symbols.

srosro, 2020-06-21:
Okay here's a new crash report with the latest DMG:

[crash_june_20.txt](https://github.com/mavlink/qgroundcontrol/files/4808771/crash_june_20.txt)

Thanks for all your work on this project.

DonLakeFlyer, 2020-06-21:
Awesome thanks. I can already see I have good info in the trace now.

Also FYI: The double connect to the two Cube Orange ports is fixed in stable and daily. But I haven't marked a new stable yet, so only daily build is available from web site. Also the new compass priority feature is also supported in newest daily.

DonLakeFlyer, 2020-06-21:
Hmm, this looks a lot like a crash sequence we've never been able to exactly track down yet. But I just reworked the code sequence around this in daily builds for performance and also should have removed the crash as well. Could you try a daily build to see if it crashes as well? That would indicate this is something different than what I was thinking about.

srosro, 2020-06-21:
Will do - want me to try a daily build with symbols or just pull what's available?

DonLakeFlyer, 2020-06-21:
Just pull what's available. I'm still working on getting symbols created correctly for all CI builds.