mavlink/qgroundcontrol #8804: QGroundControl web doc ot sure

Ankur1014, 2020-05-31:
The QGC doc throws a connection not secure error when we try to open in Chrome/Firefox. I have tried the below link 

https://dev.qgroundcontrol.com/en/

## Expected Behavior
The web doc opens without any security warning

## Current Behavior
Please describe what happens instead of the expected behavior.

## Steps to Reproduce:
Try to open the[ link](https://dev.qgroundcontrol.com/en/) in either Firefox/Chrome

[image]

DonLakeFlyer, 2020-05-31:
The DNS isn't currently pointing to the right docs either. We are working on trying to get it fixed. Hopefully soon.

closed 2020-05-31 by DonLakeFlyer

hamishwillee, 2020-06-01:
The DNS is pointing to the right docs, however the certificate provided by our hosting (github pages) is invalid. This happened with mavlink docs too, but fixed itself after a couple of days.

You can click Advanced and then go through to the pages to browse. There is no (real) risk here.