mavlink/qgroundcontrol #8808: Wayland dependency

chron0, 2020-06-02:
## Expected Behavior

Build/Run on non-wayland systems as well

## Current Behavior
 
QGC 4.x doesnt seem to work on systems that dont use wayland

## Detailed Description

I have been using QGC 3.x in the past on gentoo without any problems, build/runtime was always perfect. Trying to build 4.x fails with missing wayland dependencies. Trying to run the provided appimage breaks on missing wayland dependenies as well. Can this be solved by installing wayland libs or does it really require a running wayland system now? That would eliminate QGC from many Linux distributions that dont run wayland :/

AJRepo, 2020-07-14:
Not sure if this is a duplicate of issue #8905 ... can you try compiling with the following in qgroundcontrol.pro

`CONFIG += DISABLE_VIDEOSTREAMING `

chron0, 2020-07-16:
@AJRepo I dont think it's a duplicate, since there were wayland libs available as far as I could see in #8905. 
I dont have any wayland libs, still pure xorg based system. 

I checked out 4.0.9, added CONFIG += DISABLE_VIDEOSTREAMING to qgroundcontrol.pro and did an out of tree qmake/build run, which worked. Great! It's a shame to lose video streaming by doing this tho. However, I believe I can live with it, as I want ADSB on QGC and will have video stream running on the android tablet running the mavic proxy software. Thank you very much for the helpful suggestion to get me going again.

[image]

It would be great if I could get the video stream working as well in the future tho :)

AJRepo, 2020-07-17:
Glad to hear that worked for you! The config CONFIG += DISABLE_VIDEOSTREAMING  excludes src/VideoReceiver/* from the build which is where I think the core issue lies for both tickets.

chron0, 2020-07-20:
As this ticket got mentioned, I dont think it's necessary to keep both open. I'll close this one, as it "solved" my issue and I was able to raise my concerns about being dependent on wayland. Just re-open it if you think otherwise :)

closed 2020-07-20 by chron0