mavlink/qgroundcontrol #8812: Terrain Follow bug in QGC Stable

Antiheavy, 2020-06-06:
Not sure how I never noticed this before, but one of our guys found a bug in QGC stable (current released version) around Terrain Follow feature where you can never switch out of the Fly Tab:

100% reproducible:

1.     Plan a mission with terrain follow enabled.
2.     Upload to the vehicle
3.     Go to fly tab
4.     Everything in the fly tab works, but now you cant change tabs until you close the app

DonLakeFlyer, 2020-06-08:
I tried:
* Start PX4 SITL plane
* Boot QGC
* Go to Plan
* Add survey using template
* Use defaults for Done adjusting stuff: Basic polygon, land at vehicle location and so forth
* Change Survey to Terrain
* Go to Fly
* Go back to Plan without problem

There must be some specific other thing which triggers this.

Antiheavy, 2020-06-09:
Seems repeatable to me on multiple window's machines.  You don't even need to be connected to a vehicle,  Here is a video: https://youtu.be/YQ2Hg1cMhmQ

DonLakeFlyer, 2020-06-11:
Fixed in Stable and Daily

closed 2020-06-11 by DonLakeFlyer