mavlink/qgroundcontrol #8819: Error message when opening .plan file created on another QGC (but same version).

arnaudstephan, 2020-06-09:
Hi,

We created a .plan file into QGC 4.0.6 with a takeoff, a basic survey operation and a RTL. During creation, no hardware was connected (it's at the office, we are not connected to our usv). 

When we open this .plan file into QGC 4.0.6 on a pc connected to our USV (pixhawk cube, ardurover 4.0), the file is read and the mission appear in QGC but when we make an UPLOAD to the usv on the right top, there is an error message : MAV_PARAM3_ERROR... (sorry i didn't wrote all the message and i'm no more on my USV).

Thank you for your kind help.

[Mission.zip](https://github.com/mavlink/qgroundcontrol/files/4752391/Mission.zip)

DonLakeFlyer, 2020-06-09:
QGC needs to know the firmware and vehicle type in order to plan a mission correctly. This is because there are differences in mission command support between firmware version and even differences in mission command support between vehicle types.

You can specify the firmware and vehicle type in the Plan view Mission Start item under Vehicle Info. This can only be changed prior to adding any items to the plan though. In most cases this will automatically be set up correct by connecting a vehicle to QGC. QGC then updates these values and remembers then. But if you create a Plan before you ever connect a vehicle they can be wrong.

So the Plan you created was for PX4 Firmware and a multi-rotor vehicle. By adding a Survey into the Plan QGC added MAV_CMD_SET_CAMERA_MODE command which is supported by PX4 but not ArduPilot. The fact that ArduPilot doesn't correctly report command not supported in this case is a quirk of ArduPilot internals. Instead it reports an error which is not helpful.

Daily Builds now include a prompt at first run which asks you to specific the firmware type and vehicle type you plan to use so that you don't run into this problem.

Since you have connected a Rover by now. If you go back into Plan/Mission Start (with no plan loaded/created) and look at Vehicle Info you should see the correct settings. And if you recreate the Plan you want it should end up being correct.

So anyway, QGC is working correctly but unfortunately in a confusing way. Work is being done in new version to try to make this better.

DonLakeFlyer, 2020-06-09:
Reopening since I"m doing some testing with this for master changes to make this better. There may be a case in Stable where connecting a vehicle is not updating the settings correctly. Need to verify.

DonLakeFlyer, 2020-06-09:
I've verified stable is fine with respect to updating the fw/vehicle settings when you connect a vehicle.

closed 2020-06-09 by DonLakeFlyer

arnaudstephan, 2020-06-10:
Whoo !! Great answer, very quick and specific. Thank you so much !

Best regards