mavlink/qgroundcontrol #8839: New Feature: Altitude mode which applies to entire plan

DonLakeFlyer, 2020-06-14:
This is what the waypoint editor for a Plan looks like:
[image]

Note the big giant altitude setting control which allows you to set the altitude as well as set the altitude frame that you want to use. This same big UI shows up in an mission item which includes an altitude. The reason being, mavlink supports setting the altitude frame on each individual mission item. Hence QGC supports it as well.

The problem is that what mavlink support and what end users use in reality can be pretty far apart. Way more than the vast majorty of user create plans for which all altitudes are relatives. For these folks this big, pick your alt frame from hell ui is confusing and not needed.

In order to simplify things QGC is moving to a model where you can specific and altitude mode for an entire mission. This is done through Mission Start and defaults to relative:
[image]

Now if you add anything which has an altitude, like a waypoint you get this. Much simple and easier to understand for most users:
[image]

The options for the plan wide altitude mode are the same as what you saw in the big altitude ui:
[image]

If you pick a global alt mode other than relative you still get the little text hits about what the alt fields are doing:
[image]

If you pick "Mixed Modes" you are back to the big ass alt mode ui on each item:
[image]

So to wrap up for most users they likely won't even notice this change since they were always using relative anyway. The UI just got simpler for them.

DonLakeFlyer, 2020-06-14:
Remaining work:
* When loading an older Plan file format, grovel through it taking account of the altitude modes it uses. If all items are using the same mode then set a global mode for that plan. Otherwise set it to Mixed Mode. Currently if you load an old plan file it will always fall to Mixed Mode.
* Newly added complex items are currently not affected by this global plan alt mode.
* Current complex items support relative/absolute alts so first step is to carry through those modes to complex items.
* Survey supports Above Terrain terrain following. Carry that mode into the Survey item.
* Support Above Terrain terrain following in all the remaining complex items where approporiate.
* Lastly no complex items support creating patterns which is Terrain Frame. Add support for terrain frame to all complex items where appropriate, including global plan mode carry through.

DonLakeFlyer, 2020-06-14:
Also hopefully this helps explain the difference between Above Terrain and Terrain Frame a little bit better:
[image]

[image]

hamishwillee, 2020-06-14:
Great.

> Now if you add anything which has an altitude, like a waypoint you get this. Much simple and easier to understand for most users:

Perhaps make that field non editable by default indicating it is set mission wide - so it gets displayed as 50 (Rel) or whatever. You can perhaps make it editable with a checkbox?

Antiheavy, 2020-06-14:
This is great!

Thoughts:
- Certain patterns such as Takeoff and Landing Patterns still want to be fixed altitudes in pretty much all cases (at least the way they are currently implemented).  e.g. the various terrain follow/frame modes should not be used.  Does this new stuff still account for those special cases?

DonLakeFlyer, 2020-06-15:
> Perhaps make that field non editable by default indicating it is set mission wide - so it gets displayed as 50 (Rel) or whatever.

It seems like you are talking about the "Initial Waypoint Alt" field. Nothing here changed that functionality other than changing the label from "Waypoint Alt". It already existed. So not quite sure what you mean. It sets the initial altitude for items which set an altitude. Set it to say 50 and:
* The first takeoff item (or WP item if you have takeoff required off) you add is at 50m
* Subsequent items with alt continue to have an initial alt of whatever the previous item was if it exists. Or if not, the initial set setting.;
* Changing the initial alt value continues to prompt the user is they want to apply a new altitude to all items.
* All the same as before.

DonLakeFlyer, 2020-06-15:
> * Certain patterns such as Takeoff and Landing Patterns still want to be fixed altitudes in pretty much all cases (at least the way they are currently implemented)

Not sure why that is required. Possibly for a Takeoff in multi-rotor since that is right over Launch position so theoretically the launch alt is the highest quality value. But say for Takeoff climb out point in FW. Why wouldn't that being based on Terrain be better?

Antiheavy, 2020-06-16:
Fixed wing aircraft have climb out angles and glide slops governed by the laws of physics. They don’t care what hills may or maynot be in the way.

chris89116, 2020-06-16:
I would like to test by myself when I use the daily version of Qgroundcontrol on Android it crashes after a few seconds, it is not stable while the standard version it no problem.

remster, 2020-06-18:
I have some hopes wrt this and have two comments.
After the submission of [this](https://github.com/mavlink/qgroundcontrol/pull/8840) i still find myself to put this code in the custom plugin:
```
void CustomPlugin::postSaveToMissionJson(PlanMasterController*, QJsonObject& missionJson) {
    assert(missionJson["items"].isArray());
    assert(missionJson["plannedHomePosition"].isArray());
    QJsonArray plannedHomePosition = missionJson["plannedHomePosition"].toArray();
    assert(plannedHomePosition.size() == 3);
    missionJson.insert(
        "items",
        simpleItemsToRelativeAlt(plannedHomePosition[2].toInt(), missionJson["items"].toArray())
    );
}

QJsonArray CustomPlugin::simpleItemsToRelativeAlt(double homeAltitude, const QJsonArray& items) {
    QJsonArray result;
    for (const auto val : items) {
        assert(val.isObject());
        QJsonObject item = val.toObject();
        assert(item["type"].isString());
        QString type = item["type"].toString();
        if (type == "SimpleItem") {
            assert(item["frame"].isDouble() && item["params"].isArray() && item["command"].isDouble());
            int command = item["command"].toInt();
            if (command == MAV_CMD_NAV_WAYPOINT || command == MAV_CMD_NAV_TAKEOFF || command == MAV_CMD_NAV_LAND) {
                QJsonArray srcParams = item["params"].toArray();
                QJsonArray dstParams;
                for (int i = 0; i < srcParams.size(); i++) {
                    if (i == 6) {
                        assert(srcParams[i].isDouble());
                        dstParams.push_back(std::round(srcParams[i].toDouble() - homeAltitude));
                    } else {
                        dstParams.push_back(srcParams[i]);
                    }
                }
                int frame = item["frame"].toInt();
                if (frame == MAV_FRAME_GLOBAL) {
                    item.insert("frame", MAV_FRAME_GLOBAL_RELATIVE_ALT);
                    item.insert("params", dstParams);
                } else if (frame != MAV_FRAME_GLOBAL_RELATIVE_ALT) {
                    qCritical() << "unsupported frame " << frame;
                    assert(false);
                    exit(1);
                }
                result.push_back(item);
            } else {
                result.push_back(item);
            }
        } else if (type == "ComplexItem") {
            assert(item["TransectStyleComplexItem"].isObject());
            QJsonObject transectItem =  item["TransectStyleComplexItem"].toObject();
            assert(transectItem["Items"].isArray());
            transectItem.insert("Items", simpleItemsToRelativeAlt(homeAltitude, transectItem["Items"].toArray()));
            item.insert("TransectStyleComplexItem", transectItem);
            result.push_back(item);
        } else {
            qCritical() << "unsupported type " << type;
            assert(false);
            exit(1);
        }
    }
    return result;
}
```
to force grid surveys into relative altitude mode. 
Will implementing:
> Survey currently supports Above Terrain terrain following. Carry that mode into the Survey item.
allow me to remove this code?

DonLakeFlyer, 2020-06-19:
Instead of me having to try to figure out what this code does. Can you instead explain in detail what the problem is you solve with this code??

rtrevellyan, 2021-02-09:
On macOS Catalina 10.5.7/QGC 4.1.1, when I pick Mixed Modes in the Mission Start item, I'm unable to drop down the "big ass alt mode ui on each item" in order to pick the mode for each waypoint.

DonLakeFlyer, 2021-02-16:
> On macOS Catalina 10.5.7/QGC 4.1.1, when I pick Mixed Modes in the Mission Start item, I'm unable to drop down the "big ass alt mode ui on each item" in order to pick the mode for each waypoint.

@rtrevellyan Can you move this to a new Issue?

closed 2021-02-16 by DonLakeFlyer