mavlink/qgroundcontrol #8842: New feature:  on plan create: "Duplicate action/waypoint/etc/ below"

dkati, 2020-06-15:
Tell us a bit about the feature:
- What problem does it solve?
- Faster repeated actions

- What does it do?
- Copy the currently selected action and duplicate it below 

- What flight stacks must it work with?
- All

This feature will help user add same waypoints with different altitude...

PS
could also be replaced with an mission command like
Category/Flight Control/Change Altitude **(For copters)**

DonLakeFlyer, 2020-06-15:
[image]

Not quite as good but close.

dkati, 2020-06-15:
Replace "move" with "copy"... Actually add "copy", not replace

closed 2026-10-05 by github-actions[bot]