mavlink/qgroundcontrol #8845: Fails to run on Arch Linux (localization issue?)

DAMO238, 2020-06-16:
## Expected Behavior
Launch GUI and run.

## Current Behavior
Running causes the process to hang indefinitely and becomes unresponsive even to <C-c>. The following output is displayed when run from terminal:

```
warning: Environment variable QT_LOGGING_TO_CONSOLE is deprecated, use
QT_ASSUME_STDERR_HAS_CONSOLE and/or QT_FORCE_STDERR_LOGGING instead.
qt5ct: using qt5ct plugin

(QGroundControl:933475): Gtk-WARNING **: 17:26:55.256: Unable to locate theme engine in module_path: "adwaita",

(QGroundControl:933475): Gtk-WARNING **: 17:26:55.256: Unable to locate theme engine in module_path: "adwaita",

(QGroundControl:933475): Gtk-WARNING **: 17:26:55.257: Unable to locate theme engine in module_path: "murrine",

(QGroundControl:933475): Gtk-WARNING **: 17:26:55.257: Unable to locate theme engine in module_path: "murrine",

(QGroundControl:933475): Gtk-WARNING **: 17:26:55.257: Unable to locate theme engine in module_path: "murrine",

(QGroundControl:933475): Gtk-WARNING **: 17:26:55.258: Unable to locate theme engine in module_path: "murrine",

(QGroundControl:933475): Gtk-WARNING **: 17:26:55.258: Unable to locate theme engine in module_path: "murrine",

(QGroundControl:933475): Gtk-WARNING **: 17:26:55.258: Unable to locate theme engine in module_path: "murrine",

(QGroundControl:933475): Gtk-WARNING **: 17:26:55.258: Unable to locate theme engine in module_path: "murrine",

(QGroundControl:933475): Gtk-WARNING **: 17:26:55.258: Unable to locate theme engine in module_path: "murrine",

(QGroundControl:933475): Gtk-WARNING **: 17:26:55.258: Unable to locate theme engine in module_path: "murrine",

(QGroundControl:933475): Gtk-WARNING **: 17:26:55.258: Unable to locate theme engine in module_path: "murrine",

(QGroundControl:933475): Gtk-WARNING **: 17:26:55.258: Unable to locate theme engine in module_path: "murrine",
Settings location "/home/damien/.config/QGroundControl.org/QGroundControl.ini" Is writable?: true
Filter rules "*Log.debug=false\n"
MAVLinkLogManagerLog: MAVLink logs directory: "/home/damien/Documents/QGroundControl/Logs"
System reported locale: QLocale(English, Latin, United Kingdom) "en_GB"
Loading localization for "en_GB"
Qt localization for "en_GB" is not present
Map Cache in: "/home/damien/.cache/QGCMapCache300" / "qgcMapCache.db"
QQmlApplicationEngine failed to load component
qrc:/qml/MainRootWindow.qml:197:5: Type Drawer unavailable
file:///usr/lib/qt/qml/QtQuick/Controls.2/qmldir: plugin cannot be loaded for module "QtQuick.Controls": Cannot install type 'VerticalHeaderView' into protected module 'QtQuick.Controls' version '2'
<Unknown File>: Cannot install type 'HorizontalHeaderView' into protected module 'QtQuick.Controls' version '2'
<Unknown File>: Cannot install element 'SplitHandle' into protected module 'QtQuick.Controls' version '2'
<Unknown File>: Cannot install type 'SplitView' into protected module 'QtQuick.Controls' version '2'
<Unknown File>: Cannot install element 'Overlay' into protected module 'QtQuick.Controls' version '2'
<Unknown File>: Cannot install type 'MenuBarItem' into protected module 'QtQuick.Controls' version '2'
<Unknown File>: Cannot install type 'MenuBar' into protected module 'QtQuick.Controls' version '2'
<Unknown File>: Cannot install type 'ActionGroup' into protected module 'QtQuick.Controls' version '2'
<Unknown File>: Cannot install type 'Action' into protected module 'QtQuick.Controls' version '2'
<Unknown File>: Cannot install type 'ScrollView' into protected module 'QtQuick.Controls' version '2'
<Unknown File>: Cannot install type 'DelayButton' into protected module 'QtQuick.Controls' version '2'
<Unknown File>: Cannot install type 'ToolSeparator' into protected module 'QtQuick.Controls' version '2'
<Unknown File>: Cannot install type 'RoundButton' into protected module 'QtQuick.Controls' version '2'
<Unknown File>: Cannot install type 'MenuSeparator' into protected module 'QtQuick.Controls' version '2'
<Unknown File>: Cannot install type 'DialogButtonBox' into protected module 'QtQuick.Controls' version '2'
<Unknown File>: Cannot install type 'Dialog' into protected module 'QtQuick.Controls' version '2'
<Unknown File>: Cannot install type 'Tumbler' into protected module 'QtQuick.Controls' version '2'
<Unknown File>: Cannot install type 'ToolTip' into protected module 'QtQuick.Controls' version '2'
<Unknown File>: Cannot install type 'ToolButton' into protected module 'QtQuick.Controls' version '2'
<Unknown File>: Cannot install type 'ToolBar' into protected module 'QtQuick.Controls' version '2'
<Unknown File>: Cannot install type 'TextField' into protected module 'QtQuick.Controls' version '2'
<Unknown File>: Cannot install type 'TextArea' into protected module 'QtQuick.Controls' version '2'
<Unknown File>: Cannot install type 'TabButton' into protected module 'QtQuick.Controls' version '2'
<Unknown File>: Cannot install type 'TabBar' into protected module 'QtQuick.Controls' version '2'
<Unknown File>: Cannot install type 'SwitchDelegate' into protected module 'QtQuick.Controls' version '2'
[... 29 lines cut ...]
<Unknown File>: Cannot install type 'ButtonGroup' into protected module 'QtQuick.Controls' version '2'
<Unknown File>: Cannot install type 'Button' into protected module 'QtQuick.Controls' version '2'
<Unknown File>: Cannot install type 'BusyIndicator' into protected module 'QtQuick.Controls' version '2'
<Unknown File>: Cannot install type 'ApplicationWindow' into protected module 'QtQuick.Controls' version '2'
<Unknown File>: Cannot install type 'AbstractButton' into protected module 'QtQuick.Controls' version '2'
qt5ct: palette support is disabled
```

It appears to me that I need a certain qt module to give me localization, but I could not find such a module.

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. Install using AUR package 'qgroundcontrol'
2. Run from terminal

## System Information
When posting bug reports, include the following information
- Operating System: Arch Linux (kernel: 5.6.15-arch1-1)
- QGC Version: qgroundcontrol 4.0.8-1
- QGC build: built from source (AUR)
- Flight Controller: n/a
- Autopilot (with version): n/a

## Detailed Description
If killed with killall, trying to open a new instance of QGC causes the process to terminate due to another instance of QGC running (even though there isn't because it was killed). However, this error does not occur if the terminal is killed (with QGC as its child) and QGC is opened again.

patrickelectric, 2020-06-16:
@DAMO238 I don't have any problem with QGC here, building from source or running the appimage.

DAMO238, 2020-06-16:
I also tried the binary, same result.

Phoenox, 2020-06-17:
Heya, although I don't use QGC I stumbled across this issue while searching for these error messages. I encountered them while developing a PySide2-based application (also on Arch Linux).
I made some observations, maybe some of this can help you:
- The issue first started when I upgraded Qt to 5.15
- Launching the application with sudo (or as root) worked. Maybe this can be used as a temporary workaround.
- The actual cause for me was a user-specific Python installations that hadn't been updated. However, since QGC doesn't seem to use anything pythony, this is probably not the cause of this issue here.

DAMO238, 2020-06-17:
I tried launching with sudo (which is definitely bad practise for the long term), and got a slightly different output and it was responsive to <C-c>:
```
warning: Environment variable QT_LOGGING_TO_CONSOLE is deprecated, use
QT_ASSUME_STDERR_HAS_CONSOLE and/or QT_FORCE_STDERR_LOGGING instead.
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
qt5ct: using qt5ct plugin
QQmlApplicationEngine failed to load component
qrc:/qml/ExitWithErrorWindow.qml:16:1: Type ApplicationWindow unavailable
file:///usr/lib/qt/qml/QtQuick/Controls.2/qmldir: plugin cannot be loaded for module "QtQuick.Controls": Cannot install type 'VerticalHeaderView' into protected module 'QtQuick.Controls' version '2'
<Unknown File>: Cannot install type 'HorizontalHeaderView' into protected module 'QtQuick.Controls' version '2'
<Unknown File>: Cannot install element 'SplitHandle' into protected module 'QtQuick.Controls' version '2'
<Unknown File>: Cannot install type 'SplitView' into protected module 'QtQuick.Controls' version '2'
<Unknown File>: Cannot install element 'Overlay' into protected module 'QtQuick.Controls' version '2'
<Unknown File>: Cannot install type 'MenuBarItem' into protected module 'QtQuick.Controls' version '2'
<Unknown File>: Cannot install type 'MenuBar' into protected module 'QtQuick.Controls' version '2'
<Unknown File>: Cannot install type 'ActionGroup' into protected module 'QtQuick.Controls' version '2'
<Unknown File>: Cannot install type 'Action' into protected module 'QtQuick.Controls' version '2'
<Unknown File>: Cannot install type 'ScrollView' into protected module 'QtQuick.Controls' version '2'
<Unknown File>: Cannot install type 'DelayButton' into protected module 'QtQuick.Controls' version '2'
<Unknown File>: Cannot install type 'ToolSeparator' into protected module 'QtQuick.Controls' version '2'
<Unknown File>: Cannot install type 'RoundButton' into protected module 'QtQuick.Controls' version '2'
<Unknown File>: Cannot install type 'MenuSeparator' into protected module 'QtQuick.Controls' version '2'
<Unknown File>: Cannot install type 'DialogButtonBox' into protected module 'QtQuick.Controls' version '2'
<Unknown File>: Cannot install type 'Dialog' into protected module 'QtQuick.Controls' version '2'
<Unknown File>: Cannot install type 'Tumbler' into protected module 'QtQuick.Controls' version '2'
<Unknown File>: Cannot install type 'ToolTip' into protected module 'QtQuick.Controls' version '2'
<Unknown File>: Cannot install type 'ToolButton' into protected module 'QtQuick.Controls' version '2'
<Unknown File>: Cannot install type 'ToolBar' into protected module 'QtQuick.Controls' version '2'
<Unknown File>: Cannot install type 'TextField' into protected module 'QtQuick.Controls' version '2'
<Unknown File>: Cannot install type 'TextArea' into protected module 'QtQuick.Controls' version '2'
<Unknown File>: Cannot install type 'TabButton' into protected module 'QtQuick.Controls' version '2'
<Unknown File>: Cannot install type 'TabBar' into protected module 'QtQuick.Controls' version '2'
<Unknown File>: Cannot install type 'SwitchDelegate' into protected module 'QtQuick.Controls' version '2'
<Unknown File>: Cannot install type 'Switch' into protected module 'QtQuick.Controls' version '2'
<Unknown File>: Cannot install type 'SwipeView' into protected module 'QtQuick.Controls' version '2'
<Unknown File>: Cannot install type 'SwipeDelegate' into protected module 'QtQuick.Controls' version '2'
<Unknown File>: Cannot install type 'StackView' into protected module 'QtQuick.Controls' version '2'
<Unknown File>: Cannot install type 'SpinBox' into protected module 'QtQuick.Controls' version '2'
<Unknown File>: Cannot install type 'Slider' into protected module 'QtQuick.Controls' version '2'
<Unknown File>: Cannot install type 'ScrollIndicator' into protected module 'QtQuick.Controls' version '2'
<Unknown File>: Cannot install type 'ScrollBar' into protected module 'QtQuick.Controls' version '2'
<Unknown File>: Cannot install type 'RangeSlider' into protected module 'QtQuick.Controls' version '2'
<Unknown File>: Cannot install type 'RadioDelegate' into protected module 'QtQuick.Controls' version '2'
<Unknown File>: Cannot install type 'RadioButton' into protected module 'QtQuick.Controls' version '2'
<Unknown File>: Cannot install type 'ProgressBar' into protected module 'QtQuick.Controls' version '2'
<Unknown File>: Cannot install type 'Popup' into protected module 'QtQuick.Controls' version '2'
<Unknown File>: Cannot install type 'Pane' into protected module 'QtQuick.Controls' version '2'
<Unknown File>: Cannot install type 'PageIndicator' into protected module 'QtQuick.Controls' version '2'
<Unknown File>: Cannot install type 'Page' into protected module 'QtQuick.Controls' version '2'
<Unknown File>: Cannot install type 'MenuItem' into protected module 'QtQuick.Controls' version '2'
<Unknown File>: Cannot install type 'Menu' into protected module 'QtQuick.Controls' version '2'
<Unknown File>: Cannot install type 'Label' into protected module 'QtQuick.Controls' version '2'
<Unknown File>: Cannot install type 'ItemDelegate' into protected module 'QtQuick.Controls' version '2'
<Unknown File>: Cannot install type 'GroupBox' into protected module 'QtQuick.Controls' version '2'
<Unknown File>: Cannot install type 'Frame' into protected module 'QtQuick.Controls' version '2'
<Unknown File>: Cannot install type 'Drawer' into protected module 'QtQuick.Controls' version '2'
[... cut ...]
<Unknown File>: Cannot install type 'ButtonGroup' into protected module 'QtQuick.Controls' version '2'
<Unknown File>: Cannot install type 'Button' into protected module 'QtQuick.Controls' version '2'
<Unknown File>: Cannot install type 'BusyIndicator' into protected module 'QtQuick.Controls' version '2'
<Unknown File>: Cannot install type 'ApplicationWindow' into protected module 'QtQuick.Controls' version '2'
<Unknown File>: Cannot install type 'AbstractButton' into protected module 'QtQuick.Controls' version '2'
^C
```

DonLakeFlyer, 2020-06-19:
> * The issue first started when I upgraded Qt to 5.15

Which is not currently supported. As the dev guide says:
`Qt version: 5.12.6 (only)`

amsjavan, 2020-07-12:
I have a same problem

acxz, 2020-07-26:
> I don't have any problem with QGC here, building from source or running the appimage.

@patrickelectric can you state what commands you used to install from source? That would be very helpful.

patrickelectric, 2020-07-26:
@acxz  `qmake .. && make `

acxz, 2020-07-26:
Welp that totally worked lol.
I `clone`d
applied the following [patch](https://github.com/acxz/pkgbuilds/blob/master/qgroundcontrol/libcudata-qgc.patch)
`mkdir build && cd build && qmake .. && make`
and qgroundcontrol ran just fine.

Looks like it is something wrong with the PKGBUILD. I think it is safe to close this issue then as it is not qgroundcontrol's point of failure.

closed 2020-07-27 by DonLakeFlyer