mavlink/qgroundcontrol #8853: RC channels 15 and 16 not changing

lucasdemarchi, 2020-06-21:
On the Radio Setup screen there's no feedback for channels 15 and 16. I noticed this while implementing those channels on https://github.com/lucasdemarchi/dema-rc/

Initially I thought it was a bug on my RC sw stack, then on ArduPilot end. Finally got to check with mavproxy and I can see the values changing there. So I suspect there's something in QGC not updating the values.

I used QGC nightly to test.

DonLakeFlyer, 2020-06-21:
As far as I know ArduPilot only used to support up to 14 channels. Do you know when 15/16 support was added? And is the new support only for 16 instead of 18?

lucasdemarchi, 2020-06-22:
Number of channels depend on the board. The Linux boards always supported max 16 channels. The stm32 boards currently support 18 channels and I don't see any board overriding that. On a quick look this has always been the case since the move to Chibios for its HAL around January 2018

lucasdemarchi, 2020-06-22:
Also, SITL supports max 16 channels

DonLakeFlyer, 2020-06-22:
K, thanks. I'll have to look at why the code seems to think it needs to understand a max channels limit. The radio cal code is ancient history so it could just be stupid.

AndKe, 2020-07-03:
Current AP/AC show parameters for RC1-16 , log to dataflash RCIN,C1-14 , while some RCProtocol libraries like SBUS,  ST24, Spectrum,  contain 1-18 , but apparently go nowhere.. so yes,  there is some mess, but you can safely stop at 16.
Also, to make sure, for compatibility issues, you may check for the presence of a parameter name like RC16 before displaying ch16 as an option.

DonLakeFlyer, 2020-07-07:
So it's `RC_CHANNELS` and `RC_CHANNELS_RAW` are fighting with each other. I think way (way) back when it was something like ArduPilot only supported one of them. I think I can get rid of support for RC_CHANNELS_RAW completely now.

DonLakeFlyer, 2020-07-07:
Can you try a stable build from tomorrow to verify this is working now?

DonLakeFlyer, 2020-07-08:
Should be another daily build in about an hour

AndKe, 2020-07-09:
Current HEAD build works as expected.
Thank you, - you can close this case, and maybe look at the Disco ADB connection :)

closed 2020-07-10 by lucasdemarchi