mavlink/qgroundcontrol #8859: Compass calibration tab missing for Holybro Kakute F7 FC

bys1123, 2020-06-24:
1. Drone Configuration below, I'm sure it wired properly
[image]
2. QGC 4.0.7 Shows Compass “Setup required“ on overview page
[image]
3. But compass calibration tab is missing
[image]
only shows:
- gyro
- accel
- horizon
4. 
[image]
CLI shows compass is running well
5. QGC 3.2 works fine

DonLakeFlyer, 2020-06-25:
Whether the Compass button shows up or not is controlled by the SYS_HAS_MAG parameter. If that param exists and is set to 1 but there is no Compass button then it is a QGC problem. If the param is missing or not set to 1 then it is a firmware/hardware problem.

bys1123, 2020-06-26:
@DonLakeFlyer Thanks, SYS_HAS_MAG was 0, I will try to fix the firmware or document.

closed 2020-06-26 by bys1123

bys1123, 2020-06-26:
[image]
[image]
@DonLakeFlyer Do you have any suggestion to fix the altitude output? The baro is runing, but can't get alt information.

DonLakeFlyer, 2020-06-26:
This is related to this: https://github.com/mavlink/qgroundcontrol/issues/8851. So far I haven't heard anything from the mavlink folks.