mavlink/qgroundcontrol #8862: IOS build broken from PR 8849 onward

Noircogi, 2020-06-27:
Building under MacOS 10.15.5 with Qt 5.12.6 and XCode 11.5

Qt throws syntax errors in src/QmlControls/AppMessages.qml at line 81

The label Stable_V4.0 builds correctly with the same setup.

[image]

DonLakeFlyer, 2020-06-27:
I've never tried building with iOS with Qt Creator.  The iOS builds are always built with XCode which from just testing it a few minutes ago is working fine.

Noircogi, 2020-06-27:
How do I create the .xcodeproj for IOS without using Qt?

DonLakeFlyer, 2020-06-27:
`~/Qt/5.12.6/clang_64/bin/qmake -r ~/repos/qgc-master/qgroundcontrol.pro CONFIG+=debug -spec macx-xcode`

Then you need to set the code signing identity to "" if you don't want to sign.

Noircogi, 2020-06-27:
I verified that creating the .xcodeproj for IOS like this:
~/Qt/5.12.6/ios/bin/qmake -r ~/repos/qgroundcontrol/qgroundcontrol.pro
works for top of tree. I'm then able to build from XCode directly.

Trying to use Qt Creator to created the .xcodeproj file fails immediately for everything after PR 8849.

closed 2020-06-27 by DonLakeFlyer