mavlink/qgroundcontrol #8883: Unable to upgrade firmware on my Pixhawk 2.4.8 quads

Cattlebaron, 2020-07-06:
I find that I can't upgrade to the latest version of the firmware. Please note that I have read the manual re the upgrade procedure  many times, further I have not had problems with firmware upgrades previously.
Equipment;
iMac running OS version 10.15.5
QGroundControl version 4.0.8 (shows this version when installing the firmware)
The GUI works as expected but refuses to download the firmware the pixhawk flashes continually but the screen 'greys' out. Click the cancel button and all seems to return to normal. I find if I start QGroundControl first, select the firmware tab and then connect the Pixhawk I get the following error message:-
Error: Sync: Send Command: Invalid sync response: 0x53x 0x75
It is important to note this problem has occurred only after I upgraded the operating system on my Mac.
[image]

DonLakeFlyer, 2020-07-06:
What kind of a board is this?

Cattlebaron, 2020-07-06:
Sorry about that, the flight controller is a PIXHAWK 2.4.8

DonLakeFlyer, 2020-07-12:
Can you try a daily build. That has firmware upgrade changes with retries to get sync.

bys1123, 2020-07-12:
Did you try to change the USB cable?

Cattlebaron, 2020-07-12:
Thank's. I will try bys1123 solution first as it is the simplest and then I will tread in uncharted territory for me.

Cattlebaron, 2020-07-15:
Won't have rime to follow up for at least two weeks so I will close the issue.

closed 2020-07-15 by Cattlebaron

Cattlebaron, 2020-07-19:
Quick update, I tried a brand new cable to no avail.