mavlink/qgroundcontrol #8887: Please add support for Parrot Disco SC2 & ArduPlane.

AndKe, 2020-07-07:
This will enable QGC to be the preferred GCS for Parrot Disco with ArduPlane
@lucasdemarchi  Have done lots of great progress in making SkyController2 (the bundeled controller from now on referred to as "SC2") and Disco to run ArduPilot.

Both work fine with QGC on a laptop, by adding a automatic UDP connection to 192.168.53.1:14550 and UDP h264 stream on port 8888 )  .. but Android devices do not.

Android support is a key feature, because there is a phone/tablet holder right on the controller.
The SC2 is itself running Android, and provide a OTG USB, when connected to a phone/tablet, it present the controller and mavlink stream on the adb interface)

Please search the attached logcat for "adb"
[logcat_07-07-2020_15-13-34.txt](https://github.com/mavlink/qgroundcontrol/files/4885095/logcat_07-07-2020_15-13-34.txt)

Please make QGC handle the connection of the ADB device.

DonLakeFlyer, 2020-07-07:
It's odd that a UDP connection would not just work. There isn't any difference codewise from a udp standpoint between desktop and mobile. The issue might be that android not giving permissions to connect to that port for some reason. Do you get any errors when you set up the UDP comm link on android and connect to it? Can you get a LinkManagerLog from QGC for connecting to the UDP port.

AndKe, 2020-07-07:
This is the log with LinkManager* debug checked:
```
[D] at /home/travis/build/mavlink/qgroundcontrol/src/QGCApplication.cc:263 - "Settings location "/data/data/org.mavlink.qgroundcontrol/files/.config/QGroundControl.org/QGroundControl.ini" Is writable?: true"
[D] at /home/travis/build/mavlink/qgroundcontrol/src/QGCLoggingCategory.cc:99 - "Filter rules "*Log.debug=false\nLinkManagerLog.debug=true\nLinkManagerVerboseLog.debug=true\n""
[E] at /home/travis/build/mavlink/qgroundcontrol/src/VideoStreaming/VideoReceiver.cc:572 - "Failed because video sink is not set"
[D] at /home/travis/build/mavlink/qgroundcontrol/src/Vehicle/MAVLinkLogManager.cc:360 - "MAVLink logs directory: "/storage/emulated/0/QGroundControl/Logs""
[D] at /home/travis/build/mavlink/qgroundcontrol/src/QGCApplication.cc:363 - "System reported locale: QLocale(English, Latin, United Kingdom) "en_GB""
[D] at /home/travis/build/mavlink/qgroundcontrol/src/QGCApplication.cc:437 - "Loading localization for "en_GB""
[D] at /home/travis/build/mavlink/qgroundcontrol/src/QGCApplication.cc:443 - "Qt localization for "en_GB" is not present"
[D] at /home/travis/build/mavlink/qgroundcontrol/src/QGCApplication.cc:449 - "Error loading application localization for "en_GB""
[D] at /home/travis/build/mavlink/qgroundcontrol/src/QtLocationPlugin/QGCMapEngine.cpp:164 - "Map Cache in: "/data/data/org.mavlink.qgroundcontrol/files/QGCMapCache300" / "qgcMapCache.db""
[E] at qrc:/qml/PreFlightCheckList.qml:37 - "qrc:/qml/PreFlightCheckList.qml:37: TypeError: Cannot call method 'reset' of undefined"
[E] at qrc:/qml/QGroundControl/Controls/ToolStrip.qml:100 - "qrc:/qml/QGroundControl/Controls/ToolStrip.qml:100:21: Unable to assign [undefined] to bool"
[D] at /home/travis/build/mavlink/qgroundcontrol/src/comm/LinkManager.cc:472 - "New auto-connect UDP port added"
[E] at /home/travis/build/mavlink/qgroundcontrol/src/VideoStreaming/VideoReceiver.cc:572 - "Failed because video sink is not set"
```
I have not seen any errors.

lucasdemarchi, 2020-07-07:
@AndKe  first think we have to check is that there is a network created  between Disco and the Android device. When you connect the USB cable it can assume various roles. I think the problem to fix is how  to get it to turn on usb-ethernet.

AFAIK Parrot SDK doesn't use network but rather talk directly to the raw usb endpoint.

DonLakeFlyer, 2020-07-07:
> AFAIK Parrot SDK doesn't use network but rather talk directly to the raw usb endpoint.

Is there mavlink coming in/out of that raw usb port? I looked at parrot sdk ages ago and I thought there were some hoops to jupm through to get that in needing to call their apis.

lucasdemarchi, 2020-07-07:
> Is there mavlink coming in/out of that raw usb port? I looked at parrot sdk ages ago and I thought there were some hoops to jupm through to get that in needing to call their apis. Certainly going through udp would be much simpler.

the mavlink part from the Parrot SDK is very minimal. The "official" protocol doesn't use mavlink and rather multiplexes data using libpomp (https://github.com/Parrot-Developers/libpomp). It has been some time I looked into this so I may be wrong.

When using ArduPilot on Disco or Bebop though, we are using mavlink. And we rely on a network being create between the SC2 (the transmitter) and whatever you are connecting because this makes it much simpler and works out of the box with a desktop computer - SC2 probably detects the other side is a computer/laptop and switches its USB on-the-go port to use network. I think the debugging here would be to:

1) Is SC2 bringing up network on the USB when connecting?
2) If it is, are both devices fighting with who is running dhcp server (tethering in Android defaults to be the dhcp server)
3) If it's not, is it possible to tell android to switch to another device, or force SC2 into that mode?

Once that is done I think it should work ok with QGC, unless there is another problem like not allowing to use that port.

AndKe, 2020-07-07:
I installed Termux on android, connected SC2 when in parrrot mode and ardupilot mode, - when in parrot mode, FreeFlightPro is started automatically, when in ardupilot mode, nothing happens.  In both cases, ifconfig does not display any additional network interface.

closed 2026-10-05 by github-actions[bot]