mavlink/qgroundcontrol #8903: Resume Mission

chris89116, 2020-07-10:
I use a sprayer drone and I would like to know if it would be possible to further improve the current mission summary system.
Sometimes I have a long distance between 2 waypoints, which would save me a lot of flight time and battery saving.

Thank you for your help.

- What flight stacks must it work with : ArduPilot
- Relevant vehicle types : multirotor

DonLakeFlyer, 2020-07-11:
You're going to need to provider way more detail than that to understand what you are asking for.

Antiheavy, 2020-08-09:
@chris89116 how would you propose handing the following scenario:
1) vehicle or user commands Return Mode
2) QGC saves the vehicles current position to use as a new waypoint when it re-plans the mission
3) user then changes mode out of Return Mode for any number of reasons (maybe wants to land with stick control)
4) for some reason Return Mode is engaged again, either the user commands it or vehicle triggers it due to low battery or any other reason
5) QGC saves the position of the vehicle again at the 2nd return mode?  How should this work?

Certainly there are ways to achieve what you are asking, I've used many other ground stations that can do this, but it's complicated and in the QGC+MAVlink world there might need to be some new MAVlink functionality (I can think of ideas, but unfortunately am not someone with the skill to implement).

trhavas, 2023-01-13:
Hello. I have that same problem. Agriculture Drone . Did you solve the problem? Any progress? I would be glad if you inform me. @chris89116

trhavas, 2023-01-13:
> I don't think the best way to achieve this is a GCS development only. I
> think in order for this to be really robust part of the implementation (
> most of it ) needs to go on the autopilot firmware part. If computing this
> on the GCS it is really easy for it to get unsynced with the autopilot, but
> If we store this info in the autopilot it will be much less prone to issues.

You are absolutely right. However, it is beyond my knowledge. I made a 10lt
agricultural drone. Pretty stable. But that's my only problem.

David Sastre ***@***.***>, 13 Oca 2023 Cum, 18:23 tarihinde
şunu yazdı:

trhavas, 2023-01-13:
You are absolutely right. However, it is beyond my knowledge. I made a 10lt agricultural drone. Pretty stable. But that's my only problem.

Davidsastresas, 2023-01-14:
I think this idea is pretty cool, but we need coordination with the correspondent autopilot firmware. I really think that would be the first step to get this thing done. Can you open the correspondent issue in whatever autopilot firmware you are using?

trhavas, 2023-01-15:
Sure, here it is. Thank you for your interest. I am using Cuav Nora Flight
Control board and latest stable ArduCopter flight software. Continuing to
work from where he left off provides a great advantage.

David Sastre ***@***.***>, 15 Oca 2023 Paz, 01:48 tarihinde
şunu yazdı:

Davidsastresas, 2023-01-15:
In that case, I think you will have better chances of getting this moving posting in Ardupilot issues:

https://github.com/ArduPilot/ardupilot/issues

If you can get the attention of more users in need of this on that thread, and you manage to get it integrated, we can probably give support for it quite fast from the QGC side. Good luck!

trhavas, 2023-01-16:
Ok, I'll do that. Thanks

David Sastre ***@***.***>, 16 Oca 2023 Pzt, 01:20 tarihinde
şunu yazdı:

closed 2026-10-05 by github-actions[bot]