mavlink/qgroundcontrol #8904: No telemetry after updating PX4 with QGC

DiegoHerrera1890, 2020-07-11:
Hello, Dear community.
I updated the firmware version to the latest version of PX4.
Then, I opened the QGC and is not recognizing the PX4. I’ve seen the device manager and the driver is correctly installed.
There is a way to select the correct port in QGC?
also, I’m receiving “Error on link SIK Radio on COM3(autoconnect). Error connecting: Could not create port. The semaphore timeout period has expired”

Thank you so much

DonLakeFlyer, 2020-07-11:
Reboot your computer

DiegoHerrera1890, 2020-07-11:
I did it but doesn't work... Any suggestions?

DonLakeFlyer, 2020-07-11:
You could try turning off AutoConnect from settings and creating a manual link for it. Doesn't seem like more can be done from the QGC side. Since the error is more like a hardware problem.

DiegoHerrera1890, 2020-07-12:
Bro I've connected the PX4 to my laptop and tablet and is the same problem...

DiegoHerrera1890, 2020-07-12:
I got the answer here [updating firmware](https://nxp.gitbook.io/hovergames/userguide/qgroundcontrol/firmware#updating-the-telemetry-radio-firmware)

closed 2020-07-12 by DonLakeFlyer