mavlink/qgroundcontrol #8905: Can't run qmake on branch Stable_V4.0. "Project ERROR: Library 'wayland-cursor' is not defined."

AJRepo, 2020-07-11:
## Expected Behavior
Please tell us what should happen as detailed as possible.

qmake completes successfully 

## Current Behavior
Please describe what happens instead of the expected behavior.

Get the error 

_"Project ERROR: Library 'wayland-cursor' is not defined."_

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. mkdir build && cd build && git checkout remotes/upstream/Stable_V3.5.6 &&  qmake ../qgroundcontrol.pro
2. Build is ok
3. git checkout remotes/upstream/Stable_V4.0 &&  qmake ../qgroundcontrol.pro
4. get error "Project ERROR: Library 'wayland-cursor' is not defined."
3. git checkout master &&  qmake ../qgroundcontrol.pro
4. get error "Project ERROR: Library 'wayland-cursor' is not defined."

## System Information
When posting bug reports, include the following information
- Operating System: Ubuntu 20.04
- QGC Version: QGroundControl v4.0.9 
- QGC build: stable (also happens with latest dev commit) 
- Flight Controller: N/A
- Autopilot (with version): N/A
-QT: installed from apt repositories:  5.12.8-0ubuntu1

## Detailed Description
 I have done a grep for "wayland-cursor" and cannot find it in the 4.0 or the 3.5.6 code so I'm confused about where to solve this issue. 

## Log Files and Screenshots

Example running latest stable version
> 
> qmake ../qgroundcontrol.pro
> Project MESSAGE: Qt version 5.12.8
> Project MESSAGE: Linux build
> Project MESSAGE: QGroundControl v4.0.9
> Project MESSAGE: Release flavor
> Project MESSAGE: Using Default QtLocation headers
> Project MESSAGE: Skipping support for Pairing
> Project MESSAGE: Using MAVLink dialect 'ardupilotmega'.
> Project MESSAGE: Skipping support for Zeroconf (unsupported platform)
> Project MESSAGE: Looking for Airmap in folder "REDACTED/qgroundcontrol/libs/airmapd", variant: "Qt.5.12"
> Project MESSAGE: Taisync disabled
> Project MESSAGE: Microhard disabled
> Project MESSAGE: Including support for video streaming
> Project MESSAGE: This project is using private headers and will therefore be tied to this specific Qt module build version.
> Project MESSAGE: Running this project against other versions of the Qt modules may crash at any arbitrary point.
> Project MESSAGE: This is not a bug, but a result of using Qt internals. You have been warned!
> Project ERROR: Library 'wayland-cursor' is not defined.

AJRepo, 2020-07-11:
Should also comment that the system Ubuntu 20.04 is running wayland and has what I think are the relevant wayland libraries installed

```
$ dpkg --list | grep wayland
ii  libmirwayland-bin                          1.7.1-0ubuntu2                      amd64        Display server for Ubuntu - generator for Wayland protocol extension wrappers.
ii  libmirwayland-dev:amd64                    1.7.1-0ubuntu2                      amd64        Display server for Ubuntu - generated wrappers for Wayland
ii  libmirwayland0:amd64                       1.7.1-0ubuntu2                      amd64        Display server for Ubuntu - generated wrappers for Wayland
ii  libqt5waylandclient5:amd64                 5.12.8-0ubuntu1                     amd64        QtWayland client library
ii  libqt5waylandclient5-dev:amd64             5.12.8-0ubuntu1                     amd64        QtWayland client development files
ii  libqt5waylandcompositor5:amd64             5.12.8-0ubuntu1                     amd64        QtWayland compositor library
ii  libqt5waylandcompositor5-dev:amd64         5.12.8-0ubuntu1                     amd64        QtWayland compositor development files
ii  libuwac0-0:amd64                           2.1.1+dfsg1-0ubuntu0.20.04.1        amd64        Using wayland as a client library
ii  libuwac0-dev:amd64                         2.1.1+dfsg1-0ubuntu0.20.04.1        amd64        Using wayland as a client (development files)
ii  libva-wayland2:amd64                       2.7.0-2                             amd64        Video Acceleration (VA) API for Linux -- Wayland runtime
ii  libwayland-bin                             1.18.0-1                            amd64        wayland compositor infrastructure - binary utilities
ii  libwayland-client++0:amd64                 0.2.5-2build1                       amd64        wayland compositor infrastructure - client library C++ bindings
ii  libwayland-client-extra++0:amd64           0.2.5-2build1                       amd64        wayland compositor infrastructure - client library extra C++ bindings
ii  libwayland-client0:amd64                   1.18.0-1                            amd64        wayland compositor infrastructure - client library
ii  libwayland-cursor++0:amd64                 0.2.5-2build1                       amd64        wayland compositor infrastructure - cursor library C++ bindings
ii  libwayland-cursor0:amd64                   1.18.0-1                            amd64        wayland compositor infrastructure - cursor library
ii  libwayland-dev:amd64                       1.18.0-1                            amd64        wayland compositor infrastructure - development files
ii  libwayland-egl++0:amd64                    0.2.5-2build1                       amd64        wayland compositor infrastructure - EGL library C++ bindings
ii  libwayland-egl-backend-dev:amd64           1.18.0-1                            amd64        wayland compositor infrastructure - EGL backend development files
ii  libwayland-egl1:amd64                      1.18.0-1                            amd64        wayland compositor infrastructure - EGL library
ii  libwayland-server0:amd64                   1.18.0-1                            amd64        wayland compositor infrastructure - server library
ii  qtwayland5:amd64                           5.12.8-0ubuntu1                     amd64        QtWayland platform plugin
ii  qtwayland5-dev-tools                       5.12.8-0ubuntu1                     amd64        Qt 5 Wayland developer tools
ii  qtwayland5-doc                             5.12.8-0ubuntu1                     all          Qt 5 Wayland Compositor documentation
ii  wayland-scanner++                          0.2.5-2build1                       amd64        wayland compositor infrastructure - C++ protocol code generator
ii  waylandpp-dev:amd64                        0.2.5-2build1                       amd64        wayland compositor infrastructure - C++ development files
ii  xwayland                                   2:1.20.8-2ubuntu2.1                 amd64        Xwayland X server

```

Had no problems building, compiling, and installing qtlocation ( https://github.com/qt/qtlocation.git ) or  qtquickcontrols2  ( https://github.com/qt/qtquickcontrols2.git )

DonLakeFlyer, 2020-07-11:
At this point the only officially supported Ubuntu is 18.04. It might work on others but as you've provide, it also might now. If I had to guess Qt 5.12 does not fully support wayland.

AJRepo, 2020-07-12:
I found the commit where this error starts to be generated. 

From commit 5153a30ac49ce9d29903290f9b084490bbd8cae6 ( "Switch to statically linked qmlglsink if system does not have preinstalled one" ) and afterwards one gets the error 

I can compile against commit 2ed4a27f6fd43905c080ba5e9a4b238e69aaede1 and earlier. 

Seems related to qml or gstreamer? Will investigate more, later. 

```

$ dpkg --list | grep qml
ii  libqt5qml5:amd64                           5.12.8-0ubuntu1                     amd64        Qt 5 QML module
ii  qml-module-qtgraphicaleffects:amd64        5.12.8-0ubuntu1                     amd64        Qt 5 Graphical Effects module
ii  qml-module-qtgstreamer:amd64               1.2.0-5                             amd64        QML plugins from QtGStreamer - Qt 5 build
ii  qml-module-qtqml-models2:amd64             5.12.8-0ubuntu1                     amd64        Qt 5 Models2 QML module
ii  qml-module-qtquick-controls:amd64          5.12.8-0ubuntu2                     amd64        Qt 5 Quick Controls QML module
ii  qml-module-qtquick-layouts:amd64           5.12.8-0ubuntu1                     amd64        Qt 5 Quick Layouts QML module
ii  qml-module-qtquick-window2:amd64           5.12.8-0ubuntu1                     amd64        Qt 5 window 2 QML module
ii  qml-module-qtquick-xmllistmodel:amd64      5.12.8-0ubuntu1                     amd64        Qt 5 xmllistmodel QML module
ii  qml-module-qtquick2:amd64                  5.12.8-0ubuntu1                     amd64        Qt 5 Qt Quick 2 QML module
ii  qmlscene                                   5.12.8-0ubuntu1                     amd64        Qt 5 QML scene viewer
ii  qt5-qmltooling-plugins:amd64               5.12.8-0ubuntu1                     amd64        Qt 5 qmltooling plugins

```

```
$ dpkg --list | grep gstreamer
ii  gir1.2-gstreamer-1.0:amd64                 1.16.2-2                            amd64        GObject introspection data for the GStreamer library
ii  gstreamer1.0-alsa:amd64                    1.16.2-4                            amd64        GStreamer plugin for ALSA
ii  gstreamer1.0-clutter-3.0:amd64             3.0.27-1                            amd64        Clutter PLugin for GStreamer 1.0
ii  gstreamer1.0-gl:amd64                      1.16.2-4                            amd64        GStreamer plugins for GL
ii  gstreamer1.0-libav:amd64                   1.16.2-2                            amd64        ffmpeg plugin for GStreamer
ii  gstreamer1.0-packagekit                    1.1.13-2ubuntu1                     amd64        GStreamer plugin to install codecs using PackageKit
ii  gstreamer1.0-plugins-base:amd64            1.16.2-4                            amd64        GStreamer plugins from the "base" set
ii  gstreamer1.0-plugins-base-apps             1.16.2-4                            amd64        GStreamer helper programs from the "base" set
ii  gstreamer1.0-plugins-good:amd64            1.16.2-1ubuntu2                     amd64        GStreamer plugins from the "good" set
ii  gstreamer1.0-plugins-ugly:amd64            1.16.2-2build1                      amd64        GStreamer plugins from the "ugly" set
ii  gstreamer1.0-pulseaudio:amd64              1.16.2-1ubuntu2                     amd64        GStreamer plugin for PulseAudio
ii  gstreamer1.0-tools                         1.16.2-2                            amd64        Tools for use with GStreamer
ii  gstreamer1.0-vaapi:amd64                   1.16.2-2                            amd64        VA-API plugins for GStreamer
ii  gstreamer1.0-x:amd64                       1.16.2-4                            amd64        GStreamer plugins for X11 and Pango
ii  libgstreamer-gl1.0-0:amd64                 1.16.2-4                            amd64        GStreamer GL libraries
ii  libgstreamer-opencv1.0-0:amd64             1.16.2-2.1ubuntu1                   amd64        GStreamer OpenCV libraries
ii  libgstreamer-plugins-bad1.0-0:amd64        1.16.2-2.1ubuntu1                   amd64        GStreamer libraries from the "bad" set
ii  libgstreamer-plugins-bad1.0-dev:amd64      1.16.2-2.1ubuntu1                   amd64        GStreamer development files for libraries from the "bad" set
ii  libgstreamer-plugins-base1.0-0:amd64       1.16.2-4                            amd64        GStreamer libraries from the "base" set
ii  libgstreamer-plugins-base1.0-dev:amd64     1.16.2-4                            amd64        GStreamer development files for libraries from the "base" set
ii  libgstreamer-plugins-good1.0-0:amd64       1.16.2-1ubuntu2                     amd64        GStreamer development files for libraries from the "good" set
ii  libgstreamer-plugins-good1.0-dev           1.16.2-1ubuntu2                     amd64        GStreamer development files for libraries from the "good" set
ii  libgstreamer1.0-0:amd64                    1.16.2-2                            amd64        Core GStreamer libraries and elements
ii  libgstreamer1.0-dev:amd64                  1.16.2-2                            amd64        GStreamer core development files
ii  libqt5gstreamer-1.0-0:amd64                1.2.0-5                             amd64        C++ bindings library for GStreamer with a Qt-style API - Qt 5 build
ii  libqt5gstreamer-dev                        1.2.0-5                             amd64        Development headers for QtGStreamer - Qt 5 build
[... cut ...]
ii  libqt5gstreamerui-1.0-0:amd64              1.2.0-5                             amd64        QtGStreamerUi library - Qt 5 build
ii  libqt5gstreamerutils-1.0-0:amd64           1.2.0-5                             amd64        QtGStreamerUtils library - Qt 5 build
ii  qml-module-qtgstreamer:amd64               1.2.0-5                             amd64        QML plugins from QtGStreamer - Qt 5 build
ii  qtgstreamer-plugins-qt5:amd64              1.2.0-5                             amd64        GStreamer plugins from QtGStreamer - Qt 5 build

```

AJRepo, 2020-07-14:
Can run qmake on master (QGroundControl Development master:6cb2e1036 2020-07-09 10:08:04 -0700 ) on Ubuntu 20.04 by adding the following to  qgroundcontrol.pro

`CONFIG += DISABLE_VIDEOSTREAMING`

So I think this confirms that this issue is related to the code in src/VideoStreaming/ .

Ryanf55, 2024-01-09:
> At this point the only officially supported Ubuntu is 18.04. It might work on others but as you've provide, it also might now. If I had to guess Qt 5.12 does not fully support wayland. If so this won't work until we move to 5.16

Is there a better recommended solution now that 18.04 is EOL?

pryre, 2024-01-14:
If you're similar to the builds I've just done, you are may be relying on the system QT5 packages to build from source (rather than use QT's online installer). The error thrown is due to badly generated error message for some reason. You are likely missing the package: `qtwayland5-private-dev`

```
sudo apt install qtwayland5-private-dev
```

Someone should update the wiki instructions (perhaps in the Ubuntu section [here](https://docs.qgroundcontrol.com/master/en/qgc-dev-guide/getting_started/index.html#install-qt)) to include this. I believe I've seen a command somewhere that did a bulk installed of dependencies (perhaps for the RPi3 on discuss.ardupilot.org). Something like that would probably help in this section as well to make sure that all build dependencies are met properly.

DonLakeFlyer, 2024-01-15:
> you are may be relying on the system QT5 packages

This is not supported and hence not documented. You must always use the online installer. You can't use the OS installed packages.

closed 2024-01-15 by DonLakeFlyer

pryre, 2024-01-15:
I just realized that this was in the original author's post:
> QT: installed from apt repositories: 5.12.8-0ubuntu1

I would take back and agree that the documentation doesn't need to be listed if this isn't an issue with the supported builds. This issue should be plenty for anyone having similar troubles otherwise. Thanks Don!

DonLakeFlyer, 2024-01-15:
I'm going to be updating the master docs for the new Qt 6 instructions. I'll try to make it even more clear that you need to use the online installer.

Ryanf55, 2024-01-15:
> I'm going to be updating the master docs for the new Qt 6 instructions. I'll try to make it even more clear that you need to use the online installer.

Thanks for following up! I tried installing dependencies via CLI and got stuck. Seems like there's some better tooling on `master` for it. Please add me to the documentation PR and I'll give it a try.