mavlink/qgroundcontrol #8908: Master Crash on Takeoff deletion

FlavioTonelli, 2020-07-12:
QGC Master crash when I try to delete a Takeoff mission item while other waypoint are still planned.
To reproduce:

Start a Blank Plan mission:
Add a takeoff point 
Add a waypoint
Add another waypoint
Delete the takeoff item

The crash is related to the TerrainProfile feature. QGC is trying to access _amslTerrainHeight list when the list is empty.

DonLakeFlyer, 2020-07-12:
Need some more info:
* What is the "All Altitudes" set to
* What firmware/vehicle type in Vehicle Info?

FlavioTonelli, 2020-07-12:
I just reproduced with that configuration:
[image]
Just clicking the bin icon let QGC crash:
[image]

DonLakeFlyer, 2020-07-12:
Hmm, that's what I was trying as well.

FlavioTonelli, 2020-07-12:
I gave a try to QGC daily and I can still reproduce the problem. It seems it doesn't trigger 100 % of time but it's still quite easy to let it crash.

DonLakeFlyer, 2020-07-12:
I should be able to figure it out by inspection. There have been a bunch of crashes similar to this which I've already fixed. So I'm pretty familiar with what goes wrong. Eventually closer to stable I'll guard against these in the terrain profile code. But for now a crash is better for collected bugs.

DonLakeFlyer, 2020-07-12:
@FlavioTonelli Can you try out this change?

FlavioTonelli, 2020-07-12:
Seems promising. I can not have QGC crash on my laptop right now.
Tomorrow I will give a try on my linux workstation as well.

DonLakeFlyer, 2020-07-12:
Thanks for testing. I was able to reproduce the crash. It was caused by a stale pointer on the first item which I fixed.

FlavioTonelli, 2020-07-13:
Nice catch with that pointer!
As far as I can say the issue seems fix to me.

DonLakeFlyer, 2020-07-13:
@FlavioTonelli Thanks so much for the report

PR #8912 (2020-07-13): Plan: Fix item initialization problems leading to crashes and incorrect profile display
* Also fixes some incorrect profile display when there is no takeoff item
* Fix #8908
closed 2020-07-13 by DonLakeFlyer