mavlink/qgroundcontrol #8913: Potential Regulation / Requirements Compliance

brmyhre, 2020-07-13:
Hey all, a little background on me, I work for a company called Iris Automation, where we make a detect and avoid solution for unmanned aircraft, and I am interested in making some changes to QGroundControl to potentially help it meet regulation standards for use in BVLOS missions. 

The co-founder of my company gave a presentation at the PX4 Developer Summit this past week: https://youtu.be/aZS07EwNWAc?t=21285, and during that presentation he identified a few areas that QGroundControl may be lacking in potential requirements (slide titled “Requirements Gap”) where he indicates 3 items:

1. Continuous alerting for detected failures
     This could potentially already be met, since (depending on the autopilot) the sending of a Mavlink STATUSTEXT can be controlled, and QGroundControl will display the STATUSTEXT as long as the severity is above ERROR. So for now we might be ok with this requirement.

2. Adoption of AC 25.1322-1 (colors, terminology, etc)
     Part of this guidance outlines:
          “Visual alert indications must conform to the following color convention (§ 25.1322(e)):
               (1) Red for warning-alert indications.
               (2) Amber or yellow for caution-alert indications.
               (3) Any color except red or green for advisory alert indications.”
     This one is definitely not met, and I would love to help meet this requirement.  I'm very new to QGroundControl so I may not know the best way to handle this.  Right now, regardless of the alert, the background of the alert is yellow.  I interpret this as primarily the background color of the alert.  My initial idea was adding more alert options to the QGCPalette, such as alertBackgroundError, alertBackgroundCritical, alertBackgroundAlert, and alertBackgroundEmergency, which could then be set by default to follow that requirement, but it could also then be customizable as part of the palette.  Of course I am open to discussion on the best way to handle this, since I am definitely new to QGroundControl.

3. Visual awareness of suppressed functions (eg auditory)
     This could potentially be met as well, so I'm not sure if there is a lot to do here actually.  Once I noticed that the icon for vehicle messages changes with every STATUSTEXT I send, one might be able to argue that this visual notification is sufficient. 

So really I'd love to address item 2.  Like I said, would love to and am willing to contribute to community myself, and I’m just really starting now going through QGroundControl code so would really appreciate any direction in getting how I could get this done. And hopefully continue a relationship where I can continue to be an active contributor for the foreseeable future.

brmyhre, 2020-08-17:
Hi @DonLakeFlyer! Turns out I was wrong about #3 above as well, and we would definitely need some kind of visual indication when aural alerts are disabled.  I'm not 100% sure the best way to do that, but figured we could figure out a way.  I also have 2 additional questions:

1. Do you have a rough idea of when you think this V4.1 release might come?
2. If you could point me in the right direction to help with the color change, I could definitely do that if you feel that would help out at all.  Let me know your thoughts!

Thanks again for looking into this!

DonLakeFlyer, 2020-08-30:
I don't understand why 2 is not met? The messages dropdown uses these exact colors for messages with those three level of status: https://github.com/mavlink/qgroundcontrol/blob/master/src/ui/toolbar/MessageIndicator.qml#L34