mavlink/qgroundcontrol #8916: not able to build qgroundcontrol.pro in qt because of some error in vehicle.h

dhanusrikar, 2020-07-15:
The following template is for Issue/Bug reporting only. https://docs.qgroundcontrol.com/en/Support/Support.html#github-issues

For questions about how to use or build QGC see: http://qgroundcontrol.com/#resources

If you are using an old stable release please upgrade first and verify your issue before reporting.
----

## Expected Behavior
Please tell us what should happen as detailed as possible.

## Current Behavior
Please describe what happens instead of the expected behavior.

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1.
2.
3.
4.

## System Information
When posting bug reports, include the following information
- Operating System: Windows 10
- QGC Version: v4.0.9
- QGC build: stable
- Flight Controller: [e.g., Pixhawk I, Pixhawk Mini, Pixhawk 2, etc.]
- Autopilot (with version): [e.g., PX4 1.8.1, Ardupilot Copter 3.6.1]

## Detailed Description
Provide further details about your issue/bug.
**D:\software\qgroundcontrol-source_code\qgroundcontrol\src\Vehicle\Vehicle.h:1584: warning: C5208:** unnamed class used in typedef name cannot declare members other than non-static data members, member enumerations, or member classes to simplify migration, consider the temporary use of /Wv:18 flag with the version of the compiler with which you used to build without warnings

## Log Files and Screenshots
- [QGC Console Logs](https://docs.qgroundcontrol.com/en/Support/Support.html#reporting-bugs)
- Autopilot logs when available (post a link).
- Screenshots of QGC to help identify the current issue/bug behavior.

[image]

dhanusrikar, 2020-07-15:
i am new to both qt and qgroundcontrol, so anyways when i am trying to build the qgroundcontrol.pro i am getting the following errors. 
Thank You

DonLakeFlyer, 2020-07-18:
Are you using VS 2017? That is the only supported compiler.

dhanusrikar, 2020-07-19:
YES
[image]

DonLakeFlyer, 2020-07-19:
Hmm, that is a newer compiler than I have even though I just updated. What flavor of VS 2017 are you using? Enterprise/Community/...?

dhanusrikar, 2020-07-20:
It is community

DonLakeFlyer, 2020-08-06:
Looks like 16.6 is still in preview as far as I can tell. I don't get it when I update. How did you get 16.6?

DonLakeFlyer, 2020-08-06:
The problem is discussed here: https://developercommunity.visualstudio.com/content/problem/1034754/warning-c5208-a-c20-feature-occurs-when-compiling-1.html

closed 2020-08-06 by DonLakeFlyer