mavlink/qgroundcontrol #8920: Problems with qtandroidserialport

dakejahl, 2020-07-16:
I'm working on bringing a custom build on an android tablet back to life and back up to date with the latest release (Stable_V4.0), it was built from QGC release v3.5.2 so it's a little more than a year old. In this build, the **qtandroidserialport** is not included and the `QSerialPort::setNativeMethods()` is commented out (since this a qtandroidserialport specific function). The serial ports work, which I suppose makes sense since Qt says that QSerialPort on Android _may_ work , but the support is "not official". 

I changed the implementation to use **qtandroidserialport** and added back the  `QSerialPort::setNativeMethods()` call. Everything compiles and runs, however I cannot detect any serial ports. The call to `QSerialPortInfo::availablePorts()` fails, specifically on this call `QAndroidJniObject::callStaticObjectMethod` [see here](https://github.com/mavlink/qgroundcontrol/blob/master/libs/qtandroidserialport/src/qserialportinfo_android.cpp#L60-L63), and if you look a few lines down (line 71) `ok = false` is hit.

Any ideas on what could be going on here? Is there any reason to use the **qtandroidserialport** lib instead of QSerialPort in my case, since QSerialPort seems to be working? I am wondering more specifically if there is a performance hit, or do they both go through the JNI to access the serial ports?

I appreciate any thoughts anyone might have about this! Thanks!

DonLakeFlyer, 2020-07-17:
I don't see why you would want an android build which does serial ports differently from upstream? You've got close to 20,000 android users that prove every day that code path works.

> Qt says that QSerialPort on Android _may_ work , but the support is "not official".

I would consider that majorly not good enough to use.

dakejahl, 2020-07-17:
> prove every day that code path works.

So the issue is that it doesn't work on this device. I've combed through the diff and there's very little different that would affect this.

DonLakeFlyer, 2020-07-17:
> So the issue is that it doesn't work on this device.

What doesn't work? The existing upstream code? A specific device? Both?

dakejahl, 2020-07-17:
My serial devices were not connecting so I added some debug code and found that `QSerialPortInfo::availablePorts()` was always returning zero. So I'm not sure if this is an issue with upstream code or is a device specific issue, this is my first time developing for Android. The device is the Etlas Mobile from MMC, you can see the diff [here](https://github.com/dakejahl/qgroundcontrol/compare/mmc_forked_from_here...dakejahl:mmc_3.5.2) if you're interested.

dakejahl, 2020-07-17:
When I uninclude the qtserialportandroid library and using QSerialPort, everything works :confused:

DonLakeFlyer, 2020-07-17:
They might be running some strange version of Android which doesn't work with QGC serial port code for some random reason. Other than that pretty hard to say why it wouldn't just work.

closed 2020-07-25 by dakejahl

dakejahl, 2020-09-03:
I've been digging into this and I actually think this is a real issue, currently the **qtandroidserialport** library only handles USB devices, and the call to `availableDevicesInfo()` right [here](https://github.com/mavlink/qgroundcontrol/blob/master/libs/qtandroidserialport/src/qserialportinfo_android.cpp#L62) calls [this](https://github.com/mavlink/qgroundcontrol/blob/master/android/src/org/mavlink/qgroundcontrol/QGCActivity.java#L326-L328) which only iterates over the USB devices, and not UARTs, which appear as such: (scroll to the right and notice ttyS0 - ttyS4 and ttysWK1 - ttysWK3)
```
jake@jake:~$ adb shell
rk3399_firefly_aio_box:/ $ ls dev
__properties__ camsys_marvin1                    device-mapper graphics iep         loop-control     network_latency    ptmx      rkvdec    sw_sync ttyS3   ttysWK2 usb            watchdog   
ashmem         com.koushikdutta.superuser.daemon dri           i2c-0    iio:device0 mali0            network_throughput pts       rtc0      tty     ttyS4   ttysWK3 usb-ffs        xt_qtaguid 
binder         console                           fd            i2c-1    input       mem              null               random    rtk_btusb ttyFIQ0 ttyUSB2 tun     usb_accessory  zero       
block          cpu_dma_latency                   fscklogs      i2c-3    keychord    memcg            pmsg0              rfkill    snd       ttyS0   ttyUSB3 uhid    vendor_storage 
bus            cpuctl                            full          i2c-4    kmem        memory_bandwidth port               rga       socket    ttyS1   ttysWK0 uinput  vmdrm0         
camsys_marvin  cpuset                            fuse          i2c-9    kmsg        mtp_usb          ppp                rk3399spi stune     ttyS2   ttysWK1 urandom vpu_service
```

Do you know anyone who might be more familiar with the proper way to access UARTs on Android? I found a stackoverflow post suggesting I use `PeripheralManager` (see [this](https://developer.android.com/reference/com/google/android/things/pio/PeripheralManager)) but this would require and SDK update and some extra code in the **QGCActivity.java**

dakejahl, 2020-09-03:
Also I should mention, the only way Android can properly detect USB devices is if the VID/PID is added to the list of known devices, which seems a bit odd.
https://github.com/mavlink/qgroundcontrol/blob/master/android/src/com/hoho/android/usbserial/driver/UsbId.java