mavlink/qgroundcontrol #8932: Advanced Tuning for VTOL airframe (in Multirotor mode)

saengphet, 2020-07-23:
Hello, I have experienced tuning multirotor attitude controller using Advanced Tuning before (using graph and adjust PID gains). This feature reduces time-consuming in tuning multirotor sinificantly. However, It could not be used to tune VTOL in multirotor mode because there are no Advanced Tuning Tab. Is it possible to implement Advanced Tuning for VTOL airframe using graph and adjust PID gains in the same way as multirotor airframe?

reference page:
https://docs.qgroundcontrol.com/en/SetupView/tuning_px4.html

saengphet, 2020-08-02:
I have already implemented this feature (VTOL in MC mode PID tuning) by myself. Thanks

closed 2020-08-02 by saengphet

DonLakeFlyer, 2020-08-02:
Submit a pull!

saengphet, 2020-08-03:
> Submit a pull!
Hi, we are testing this feature to ensure that everything is ok after that I will submit PR as your recommedation. Thank you !!