mavlink/qgroundcontrol #8934: Crash bug in UDPLink::_writeBytes()

dogmaphobic, 2020-07-24:
I've been chasing this all day and no definitive answer. It takes a while to get it to happen on my regular computer but it is quite easy to reproduce on Android. But as it depends on sending lots of data (Joystick), not easy to test unless you have an Android device with a joystick.

Just launch QGC, connect to anything (even Gazebo) and make sure you have a joystick and it is enabled. Once connected, you don't need to do anything. Just wait. On Android, it will crash within a minute. On desktops, it may take a while. It's a game of chance. No hard rule on when it happens (if you really feel like it, switching the Joystick code to send messages at 200Hz speeds up the crash interval).

The crash is here:
https://github.com/mavlink/qgroundcontrol/blob/master/src/comm/UDPLink.cc#L189

`target` is bogus. Not NULL but bogus.

The call comes from here:
https://github.com/mavlink/qgroundcontrol/blob/master/src/comm/UDPLink.cc#L182

When you hit the crash, the contents of `_sessionTargets` is valid. Note that `_sessionTargets` is only written to at the very beginning of the connection (literally when QGC receives the first packet from a vehicle). I don't fully grasp the nuances of the voodoo behind these c++11 constructs. The doc says the above gets a copy of the pointer. It looks like instead of:
```
for(UDPCLient* target: _sessionTargets) {
```
It should be:
```
for(auto& target: _sessionTargets) {
```
What I can tell is if I switch to the explicit:
```
for(int i = 0; i < _sessionTargets.count(); i++) {
    UDPCLient* target = _sessionTargets[i];
```
the crash goes away.

Before pushing a PR, I would like to hear more about those who actually eat c++11 for breakfast. These sort of things seem to be all over the place in the code base.

DonLakeFlyer, 2020-07-25:
Seems to me like your change is just rearranging the deck chairs on a possible stack corruption or something. Code one way and the variable is on the stack, code another and the variable is in a register maybe. Or the addition of `int i` moved the stack around as well. Linux debug should tell you if stack is walked over. OSX/Xcode you can build/run with the memory check thingy which was how I figured out this one: https://github.com/mavlink/qgroundcontrol/pull/8885/files

DonLakeFlyer, 2020-07-25:
Crash is caused by the hidden iterators which operate behind the scenes when using C++ range_for. Both read and write bytes can come in on different thread and try to use iterators on the same object at the same time and then boom.

DonLakeFlyer, 2020-07-25:
> Crash is caused by the hidden iterators which operate behind the scenes when using C++ range_for.

Not so sure yet. Still looking.

DonLakeFlyer, 2020-07-26:
So the answer is to never use C++ range_for against any Qt list in the QGC codebase. There is major quirkiness with respect to how these work and how implicit sharing and detach on Qt objects cause problems with the iterators.