mavlink/qgroundcontrol #8937: QGC IOS App update

Yannicflight, 2020-07-27:
Hello QGC Developers.

I found that the QGC version in the Apple store for IOS devices is still at version 3.4.3 which is released approximately a year ago.
Since then there where numerous improvements for the PC, Mac and Android versions which are now at 4.0.6. Especially the newer Gstreamer implementation works much better and less laggy then the one used in QGC v3.4.3

I understand that due to apple being in the loop releases for IOS are a bit harder to do and that the versions will not be as current as they will be on android devices, but this would be a big improvement in the usability of the APP.

I already posted this on the QGC forum but without a response so far: https://discuss.px4.io/t/qgc-ios-update/17781

DonLakeFlyer, 2020-07-27:
I'm hoping to have time to get a new version up soon. I've taken over the process from someone else so it's taken some time for me to learn how to do it.

DonLakeFlyer, 2020-07-28:
Submitted and waiting on Apple approval

Yannicflight, 2020-07-29:
Great news! Thank you @DonLakeFlyer  for the quick response and actions. Looking forward to testing it!

DonLakeFlyer, 2020-07-29:
Still in hand to hand combat with approval folks

DonLakeFlyer, 2020-07-30:
4.0.10 is available in the store

closed 2020-07-30 by DonLakeFlyer