mavlink/qgroundcontrol #8940: Plan: Transition from existing plan to different vehicle does not work correctly

DonLakeFlyer, 2020-07-27:
Example problem:
* Start QGC 
* Connect to a vehicle with a Plan on it
* Disconnect that vehicle
* Connect to another vehicle with a different or no Plan (this could even be the same vehicle which had the plan on it futzed with in between connects)
* Plan view shows stale plan from original vehicle connect

There are many different incarnations of cases where things are incorrect. Above is just one example. The question is what to do when the Plan being shown is stale with respect to either the vehicle it was associated with disconnecting or a new active vehicle coming into play.

Here is how it should work for the various transitions:
* Plan view is showing a plan which is not dirty
  * New vehicle becomes active: Plan view should change to show the Plan from that vehicle
  * Current active vehicle disconnects leaving no active vehicle: Plan view should clear to no plan
* Plan view is showing a plan which is dirty:
  * For all above cases prompt the user to save to plan to disk
  * When the user saves/cancels take the action associated with Plan not dirty above

This can be accomplished by modifying `PlanMasterController::_activeVehicleChanged` for the Plan view case:
* If the plan is not dirty: `removeAll` if needed and then `_showPlanFromManagerVehicle`
* If the plan is dirty signal something like `savePlanPriorToVehicleChange`
* Plan view:
  * Add handler for `savePlanPriorToVehicleChange`
  * Prompt the user to use save to disk or cancel to toss changes
  * After save or cancel call a new method in PlanMasterController 'showPlanFromManagerVehicle' which correctly handles the newly offline case (removeAll) and newly online case (`_showPlanFromManagerVehicle`)

Not trivial but still fairly simple.

DonLakeFlyer, 2020-07-27:
Note that this changes the following user model sequence which may currently be in use:
* Create an plan while offline (not connected to vehicle)
* Connect to a vehicle
* Upload the plan you created to the vehicle

Not sure how important that is. Maybe @antiheavy should comment.

DonLakeFlyer, 2020-07-27:
All of this always ends up being way more complicated than you think it is.

DonLakeFlyer, 2020-07-27:
I need to put some more though into this. My proposal above I think is going to be too annoying for some use cases.

DonLakeFlyer, 2020-07-27:
I removed the here is how it should work section from the first comment since I realized it was crap once I started working on this. I'll have a detailed post on how it works soon as I'm done.

DonLakeFlyer, 2020-07-27:
> All of this always ends up being way more complicated than you think it is.

And yes, this statement ended up being truer than I originally thought.

DonLakeFlyer, 2020-07-27:
To explain how it will work you need to understand the concept of the Plan View running "online" or "offline". Online means that the Plan View has a real vehicle associated with it, you can save/load to disk or upload/download to the vehicle. Offline means that the Plan View does not have a connected vehicle associated with it. You can only save/load from disk. 

The transitions from offline to online and back are what cause all the complexity with the user model. The contents of the Plan View itself also add to the various states to manage. It can be: empty, clean plan, dirty plan. Dirty means the plan have changed since last load/save/download/upload. Clean means there is a plan but there are no unsaved changes.

So here is how it works explained against these states:

* Offline -> Online
  * Plan view empty or clean - Plan view will change to show the Plan from the new vehicle
  * Plan view dirty - Plan view will show prompt asking the user what to do
* Online -> Offline
  * Plan view empty or clean - Plan view will be cleared
  * Plan view dirty - Plan view will show prompt asking the user what to do
* Online -> Online (active vehicle change)
  * Plan view empty or clean - Plan view will change to the plan from the new active vehicle
  * Plan view dirty - Plan view will show prompt asking the user what to do

The two prompts you get are based on whether the end state of the transition is online or offline:
[image]

[image]

DonLakeFlyer, 2020-07-27:
@hamishwillee You might want to look at the wording here to see what you think.

DonLakeFlyer, 2020-07-28:
The other quirkiness with this is that those popups show up in the current view you are in. Not just in the Plan view. We don't currently have a mechanism for dialogs which are parented to a specific view. I figure since this only happen in certain cases that it's good enough. If it's determined to be a real problem I can put some more effort into it to support parenting to the Plan View only.

closed 2020-07-28 by DonLakeFlyer

Antiheavy, 2020-07-28:
Sorry for the delayed reply. My view on this is pretty much what you’ve been discovering. It’s complicated and hard. The approach I’ve seen work best in many other ground station softwares is when the onboard mission is shown as a ghost (translucent) or different color. CRC or other methods of knowing if the vehicle and ground are out of sync are typically used.