mavlink/qgroundcontrol #8942: Android text to speech audio output problem

moreba1, 2020-07-28:
Hi

There is an issue in audio output of QGC. My android phone and tablet have no audio output in QGC . so in my opinion qgc dont't work  properly with some android devices.

similar problem: 
https://discuss.px4.io/t/qgc-have-no-audio-output-in-android/9315
https://discuss.ardupilot.org/t/help-needed-on-qgc-audio-speech-alerts-for-android/58205

DonLakeFlyer, 2020-07-28:
Not sure. I think I've heard people say they tweaked some settings on their devices and got it to work.

moreba1, 2020-07-28:
> settings

wich settings ? diong media sound setting ?

DonLakeFlyer, 2020-07-28:
No idea. I'd ask on slack and/or qgc forums.

DonLakeFlyer, 2020-07-28:
Let me know if you figure out any details

moreba1, 2020-09-05:
I realized the problem
English must be selected for the text-to-speech language.

closed 2020-09-05 by moreba1