mavlink/qgroundcontrol #8953: Display loiter circle for planes in mission planing

moreba1, 2020-07-30:
In mission planning, it is better that we must see the actual path of flight to avoid the collision. also when plane exit tangent frome loiter, the direction of the path is different from that see in UI. this feature is needed for VTOLs and planes.

DonLakeFlyer, 2020-07-30:
I think this might be doable in a generic fashion. That said though the flight path shown in Plan should be taken with a huge grain of salt. You need real vehicle flight dynamic information to know what will really happen.

hamishwillee, 2020-07-30:
@DonLakeFlyer FWIW I've recently updated the docs on what the LOITER commands are doing [here](https://mavlink.io/en/services/mission.html#loiter_commands). Provided the heading =1 is set and xtrack = 0 or 1 the flight is fairly predictable on PX4 (see images at link).

I don't know if ArduPilot implements the heading yet but it is in PX4 master. Any value for xtrack other than 0 or 1 is defined in mavlink but FWIW not likely to be implemented in PX4 or ArduPilot.

Antiheavy, 2020-08-03:
Related; https://github.com/mavlink/qgroundcontrol/issues/6721

moreba1, 2021-01-02:
My proposed Graphics:
[image]

DonLakeFlyer, 2021-01-02:
How does the loiter entry path work? It doesn't enter from the tangent to the loiter circle? So line should go from 4 to appropriate tangent on loiter circle?

moreba1, 2021-01-02:
Entry Path is not always same. It is related to Navigation tuning of plane and other parameters.