mavlink/qgroundcontrol #8958: Add Support for MAVLink2 Message Signing

jonas-koeritz, 2020-08-01:
Tell us a bit about the feature:
- What problem does it solve?
Integrity checks and authentication of GCS and Vehicle prevent hijacking of a vehicle via MAVLink telemetry radio.

- What does it do?
Prevent rogue ground stations from forcing a vehicle to land or upload malicious missions etc.

- What flight stacks must it work with? (All, PX4, ArduPilot)
All

- Relevant vehicle types? (All, multirotor, fixed-wing, VTOL, submarine, etc.)
All

- Are there other systems that have this feature?
MAVLink2 contains this feature (https://mavlink.io/en/guide/message_signing.html) and Mission Planner seems to support it.

- What communication/integration standards does the feature rely on (e.g. MAVLink commands etc.)
MAVLink2

TByte007, 2022-07-18:
Is this implemented or not after 2 years :) And If I want to implement it where should I look (which part of the code) ?

holden-zenith, 2022-07-19:
It was but then it wasn't. I don't think it ever got [merged](https://github.com/mavlink/qgroundcontrol/pull/5582).

holden-zenith, 2022-07-20:
PX4 doesn't support signing. It's unlikely signing or anything else will get added anymore at this point. It's trivial to implement if you want to do it yourself for your own use.

TByte007, 2022-07-20:
> PX4 doesn't support signing. It's unlikely signing or anything else will get added anymore at this point. It's trivial to implement if you want to do it yourself for your own use.

Unlikely in PX4 or qgroundcontrol and why ? The project it dead or for some other reason ?

holden-zenith, 2022-07-20:
Unlikely to be added in qgroundcontrol, I believe it's feature frozen.

HironariNakama, 2023-04-10:
> PX4 doesn't support signing. It's unlikely signing or anything else will get added anymore at this point. It's trivial to implement if you want to do it yourself for your own use.

How can I implement it myself for my own use?

tenchirocom, 2023-04-15:
@HironariNakama I have implemented signing for a companion computer library and use it through MissionPlanner. However, I often use qGroundControl for flying and have been looking at it for some GCS add-ons and was trying to figure out how to turn it on, when I stumbled across this thread. Basically, you just need to compute a checksum and md hash. Then add it into the message appropriately. I have yet to build qgc, but assuming there is a layer point for communications, the messages can just be tagged there. Maybe someone has verified, but if signing is activated through MissionPlanner, should be able to see telemetry in QGC, but not change parameters or send commands. If you want to turn it on, there'd need to be a very basic ui. The MP ui is super basic. If you are interested to collaborate on this in some way, send me a message. The biggest question I have is how to do it in a way it can be easily merged with future versions. Maybe someone has experience or pointers about that.

lida2003, 2023-04-26:
does latest 4.2.6 support mavlink2 signing protocol? As I didn't find authkey settings.

tenchirocom, 2023-04-26:
ArduCopter autopilot supports it. You can activate through MissionPlanner. I have not found a way to manage it through qGroundControl however.

lida2003, 2023-04-27:
> ArduCopter autopilot supports it. You can activate through MissionPlanner. I have not found a way to manage it through qGroundControl however.

yeah,  i have activate in mp. But i don't know where to config in qgc. googled but no guides, only this request.

tenchirocom, 2023-04-27:
I haven't tested yet, but once activated on the aircraft through MP, QCG without signing support should still display the telemetry data. However, the aircraft should refuse to respond to any commands, including the request for parameters. Not sure if that is enough for your purposes. For my scenarios, parameter configuration in the field is important. So is configuring and starting a mission. Given the issues, I assume most folks just use an encrypted radio link and don't worry about it?

lida2003, 2023-04-27:
Well, what I have met is, QGC has trouble in vechicle setup page.  I'm NOT sure about the other functions in QGC, and I will never try to fly with MAVLink2-QGC.

hamishwillee, 2023-11-01:
QGC does support MAVLink 2. It just doesn't support signing. 
The stuff that needs to be implemented is https://mavlink.io/en/mavgen_c/message_signing_c.html
There are two partial implementations in QGC that never went in, which would be a good place to understand the needed code.

tenchirocom, 2023-11-01:
Thank you for this code pointer. I will take a look.

With what I've experienced implementing signing on the aircraft side, I'm wondering if anyone really uses it. Mission Planner works well enough. But between the companion computers and fcs there are a few idiosyncrasies that have me wondering if mavlink signing is the way to go or maybe the best approach is to just secure the link. Then there'd be privacy too.

mjemv, 2024-02-27:
Hi @hamishwillee All, any sample code to look at in send and recv side?

PR #11596 (2024-07-27): MAVLink: Add Signing
Todo: Probably should disable the enable switch until a valid key is entered? 
Closes #8958

[image]

[image]
closed 2024-07-27 by HTRamsey