mavlink/qgroundcontrol #8974: Planning UI issues in daily build version

moreba1, 2020-08-08:
Hi

some weak points are in the daily build version and it is better that be corrected.

1- The lines are very thin.
2- Height bar does not scale correctly when survey terrain follow activated.
3- It is better that under terrain alt showed filled in height graph.

[image]

DonLakeFlyer, 2020-08-08:
> Height bar does not scale correctly when survey terrain follow activated.

Can you attach a plan file where this is incorrect?

moreba1, 2020-08-08:
Plan with terrain follow 
[image]

Plan without terrain follow
[image]

[plans.zip](https://github.com/mavlink/qgroundcontrol/files/5046155/plans.zip)

DonLakeFlyer, 2020-11-20:
> Height bar does not scale correctly when survey terrain follow activated.

This should be fixed now. Enter a separate issue if there is still a bug here.

moreba1, 2020-11-22:
> This should be fixed now. Enter a separate issue if there is still a bug here.

currently follow terrain removed from survey and AGL alt setting for the waypoint. also, the height bar widget shows the altitude of waypoints lower than the terrain ground altitude.

[image]

DonLakeFlyer, 2020-11-23:
Can you move this to a new issue?

moreba1, 2020-11-23:
> Can you move this to a new issue?

https://github.com/mavlink/qgroundcontrol/issues/9207