mavlink/qgroundcontrol #8978: Cannot find the androiddeploy Json file.

SpringSmell, 2020-08-10:
I'm selected android phone run project with ARMv7. 
encounter a problem:
```
Initializing deployment to Android device/simulator
Deploying to 621QEBPS2WU59
Cannot find the androiddeploy Json file.
Error while building/deploying project qgroundcontrol (kit: Android for armeabi-v7a (Clang Qt 5.12.9 for Android ARMv7))
When executing step "Deploy to Android device"
```

SpringSmell, 2020-08-10:
use QT version
`Qt Creator 4.12.2
Based on Qt 5.14.2 (MSVC 2017, 32 bit)

Built on Jun 2 2020 01:31:56

From revision 33c5ff216a`

SpringSmell, 2020-08-10:
I'm use QT version
```
Qt Creator 4.12.2
Based on Qt 5.14.2 (MSVC 2017, 32 bit)

Built on Jun 2 2020 01:31:56

From revision 33c5ff216a
```

SpringSmell, 2020-08-10:
log info 
```
Project MESSAGE: Qt version 5.12.9
Project MESSAGE: Android Arm 32 bit build
Project MESSAGE: Found ccache, enabling
Project MESSAGE: Daily Build
fatal: Not a valid object name HEAD
fatal: ambiguous argument 'HEAD': unknown revision or path not in the working tree.
Use '--' to separate paths from revisions, like this:
'git <command> [<revision>...] -- [<file>...]'
fatal: Needed a single revision
fatal: ambiguous argument 'HEAD': unknown revision or path not in the working tree.
Use '--' to separate paths from revisions, like this:
'git <command> [<revision>...] -- [<file>...]'
Project MESSAGE: QGroundControl Development HEAD: 
Project MESSAGE: Debug flavor
Project MESSAGE: Using Default QtLocation headers
Project MESSAGE: Skipping support for Pairing
Project MESSAGE: Using MAVLink dialect 'ardupilotmega'.
Project ERROR: MAVLink folder does not exist at 'libs/mavlink/include/mavlink/v2.0'! Run 'git submodule init && git submodule update' on the command line.
[Inexact] Cannot read C:/Qt/Project/qgroundcontrol-master/user_config.pri: No such file or directory
```

closed 2020-08-14 by DonLakeFlyer

ugrrelaise, 2021-06-22:
I have encountered with this problem. Could you solved it ? Help pls.