mavlink/qgroundcontrol #8981: [Bug] Plan tab altitude plot only plots in Meters

kosakwayne, 2020-08-12:
Using DailyBuild from August 12th

It seems that the plan tab "Height AMSL()" values are plotted in meters, regardless of the axis label or unit set in Qgroundcontrol settings

mission planned in Bozemon Montana with base units in meters
[image]

Changed units in Qgroundcontrol to feet. Restarted application
Same location for mission planning: Bozemon montana with base units in feet
[image]

The plot values are very similar, but should be different by a factor of 3.3. Bozeman's alttitude is ~1460m, showing that using the base units of Meters yields correct results

DonLakeFlyer, 2020-08-16:
@kosakwayne There should be a new daily in about an hour. Can you verify the fix?

DonLakeFlyer, 2020-08-16:
Hold off on that. I found another problem.

DonLakeFlyer, 2020-09-15:
This should be fixed now

closed 2020-09-15 by DonLakeFlyer