mavlink/qgroundcontrol #8986: Pitch trimming screen for fixed wing UAV

jstrebel, 2020-08-14:
QGoundControl has a very practical tool that simplifies the setting of the PID controller for a UAV.
For fixed wing UAV's it would be very useful if a similar tool would be available for trimming the elevator.
Function 1) Setting the trim correction value of the elevator at Vmin, Cruise Speed and Vmax.
I understand the following parameters would be desirable: FW_PSP_OFF, FW_DTRIM_P_VMIN and FW_DTRIM_P_VMAX
Function 2) Adjustment of the elevator trim correction value for different flap settings.
Parameters: FW_DTRIM_P_FLPS optional FW_DTRIM_R_FLPS

On this "Tuning Screen" the following values should be displayed during this process:
Airspeed, Groundspeed ,Altitude or sink/rising rate.

Jakob

jstrebel, 2020-08-25:
Hello, I am not sure if my request is not clear formulated, or am I alone with the opinion that Fixed Wing trimming could be made much user friendly, like the PID Tuning.
@DonLakeFlyer  could you please shed some light on this? Thank you. Jakob

dagar, 2020-12-29:
Interesting idea. Related to this I'd really like to do a much better exposing the mixer setup in general. Configuring each output intuitively, making it easy to extend, etc.

closed 2026-10-05 by github-actions[bot]