mavlink/qgroundcontrol #9005: Add Azerbajani translation to QGC

hamishwillee, 2020-08-19:
The language has already been added to Crowdin

farid002, 2020-08-24:
[translations.zip](https://github.com/mavlink/qgroundcontrol/files/5116406/translations.zip)
Azerbaijani translation files due to be uploaded to Crowdin project

hamishwillee, 2020-08-24:
@DonLakeFlyer @dogmaphobic Can one of you add those translations to crowdin - I presume created "offline". I haven't done it before and don't want to screw anything up.

DonLakeFlyer, 2020-08-24:
Neither have I. @farid002 The new language has been added to Crowdin are you able to get these in there yourself now?

farid002, 2020-08-24:
Yes, it is open. The thing is that we already have translated QGC on our local machines. Previously I sent those zipped .ts files. 

Now, not to translate whole project again, can you just change current .ts files of Azerbaijani into the one I sent as .zip, please?

farid002, 2020-08-24:
On Crowdin

DonLakeFlyer, 2020-08-25:
I look at the files in the zip file. Nothing in them in translated as far as I can see.

hamishwillee, 2020-08-26:
Ditto - there might be something, but the zip file in https://github.com/mavlink/qgroundcontrol/issues/9005#issuecomment-678965209 appears to be just the original files on quick scan.

farid002, 2020-08-26:
[translations.zip](https://github.com/mavlink/qgroundcontrol/files/5129086/translations.zip)
Sorry for incorrectness, here is the translated ones zipped

hamishwillee, 2020-08-27:
@farid002 @DonLakeFlyer IMported - fantastic job
[image]

Farid, note that these are not approved. Normally we have a translator and a proofreader or two proofreaders. The point being, it is good if there are two people reviewing each other's work. I'd happily make you proofreader, but is there someone else you can get to join the project too so you can review each other's work?

closed 2020-08-27 by hamishwillee

farid002, 2020-08-28:
@hamishwillee  Yes, I have a team mate who will also join to the Crowdin project. Thank you @hamishwillee @DonLakeFlyer  for your support!