mavlink/qgroundcontrol #9006: PX4 compass calibration not showing rotation configuration for all external mags

dagar, 2020-08-19:
[image]

On a system with 4 mags (3 external).

``` Console
nsh> listener sensor_mag

TOPIC: sensor_mag 4 instances

Instance 0:
 sensor_mag_s
        timestamp: 129496038  (0.032901 seconds ago)
        timestamp_sample: 129495158  (880 us before timestamp)
        device_id: 396825 (Type: 0x06, I2C:3 (0x0E)) 
        x: -0.1980
        y: 0.2490
        z: 0.0750
        temperature: nan
        error_count: 0
        is_external: False

Instance 1:
 sensor_mag_s
        timestamp: 129505609  (0.024651 seconds ago)
        timestamp_sample: 129504776  (833 us before timestamp)
        device_id: 396833 (Type: 0x06, I2C:4 (0x0E)) 
        x: 0.1560
        y: -0.3000
        z: 0.7710
        temperature: nan
        error_count: 0
        is_external: True

Instance 2:
 sensor_mag_s
        timestamp: 129504978  (0.026603 seconds ago)
        timestamp_sample: 129504145  (833 us before timestamp)
        device_id: 396809 (Type: 0x06, I2C:1 (0x0E)) 
        x: -0.1140
        y: 0.0960
        z: 0.2520
        temperature: nan
        error_count: 0
        is_external: True

Instance 3:
 sensor_mag_s
        timestamp: 129496099  (0.036775 seconds ago)
        timestamp_sample: 129494768  (1331 us before timestamp)
        device_id: 335393 (Type: 0x05, I2C:4 (0x1E)) 
        x: -0.1530
        y: -0.4209
        z: 0.5074
        temperature: 32.2500
        error_count: 0
        is_external: True
```

``` Console
nsh> param show CAL_MAG*
Symbols: x = used, + = saved, * = unsaved
x + CAL_MAG0_ID [78,104] : 396825
x + CAL_MAG0_PRIO [79,105] : 50
x + CAL_MAG0_ROT [80,106] : -1
x + CAL_MAG0_XCOMP [81,107] : 0.0000
x + CAL_MAG0_XODIAG [82,108] : 0.0046
x + CAL_MAG0_XOFF [83,109] : -0.0902
x + CAL_MAG0_XSCALE [84,110] : 0.9990
x + CAL_MAG0_YCOMP [85,111] : 0.0000
x + CAL_MAG0_YODIAG [86,112] : 0.0060
x + CAL_MAG0_YOFF [87,113] : 0.1021
x + CAL_MAG0_YSCALE [88,114] : 1.0319
x + CAL_MAG0_ZCOMP [89,115] : 0.0000
x + CAL_MAG0_ZODIAG [90,116] : -0.0959
x + CAL_MAG0_ZOFF [91,117] : -0.1957
x + CAL_MAG0_ZSCALE [92,118] : 1.0441
x + CAL_MAG1_ID [93,119] : 396833
x + CAL_MAG1_PRIO [94,120] : 75
x + CAL_MAG1_ROT [95,121] : 6
x + CAL_MAG1_XCOMP [96,122] : 0.0000
x + CAL_MAG1_XODIAG [97,123] : -0.0004
x + CAL_MAG1_XOFF [98,124] : 0.2117
x + CAL_MAG1_XSCALE [99,125] : 0.9619
x + CAL_MAG1_YCOMP [100,126] : 0.0000
x + CAL_MAG1_YODIAG [101,127] : -0.0267
x + CAL_MAG1_YOFF [102,128] : -0.0273
x + CAL_MAG1_YSCALE [103,129] : 1.0337
x + CAL_MAG1_ZCOMP [104,130] : 0.0000
x + CAL_MAG1_ZODIAG [105,131] : 0.0409
x + CAL_MAG1_ZOFF [106,132] : 0.2399
x + CAL_MAG1_ZSCALE [107,133] : 0.9946
x + CAL_MAG2_ID [108,134] : 396809
x + CAL_MAG2_PRIO [109,135] : 75
x + CAL_MAG2_ROT [110,136] : 0
x + CAL_MAG2_XCOMP [111,137] : 0.0000
x + CAL_MAG2_XODIAG [112,138] : -0.0077
x + CAL_MAG2_XOFF [113,139] : 0.0619
x + CAL_MAG2_XSCALE [114,140] : 0.9855
x + CAL_MAG2_YCOMP [115,141] : 0.0000
x + CAL_MAG2_YODIAG [116,142] : -0.0387
x + CAL_MAG2_YOFF [117,143] : 0.0373
x + CAL_MAG2_YSCALE [118,144] : 0.9996
x + CAL_MAG2_ZCOMP [119,145] : 0.0000
x + CAL_MAG2_ZODIAG [120,146] : -0.0183
x + CAL_MAG2_ZOFF [121,147] : -0.1019
x + CAL_MAG2_ZSCALE [122,148] : 1.0187
x + CAL_MAG3_ID [123,149] : 335393
x + CAL_MAG3_PRIO [124,150] : 75
x + CAL_MAG3_ROT [125,151] : 6
x + CAL_MAG3_XCOMP [126,152] : 0.0000
x + CAL_MAG3_XODIAG [127,153] : 0.0119
x + CAL_MAG3_XOFF [128,154] : -0.1288
x + CAL_MAG3_XSCALE [129,155] : 0.9870
x + CAL_MAG3_YCOMP [130,156] : 0.0000
x + CAL_MAG3_YODIAG [131,157] : -0.0221
x + CAL_MAG3_YOFF [132,158] : -0.1864
x + CAL_MAG3_YSCALE [133,159] : 1.0063
x + CAL_MAG3_ZCOMP [134,160] : 0.0000
x + CAL_MAG3_ZODIAG [135,161] : -0.0013
x + CAL_MAG3_ZOFF [136,162] : 0.0119
x + CAL_MAG3_ZSCALE [137,163] : 0.9897
x   CAL_MAG_COMP_TYP [138,164] : 0
x   CAL_MAG_ROT_AUTO [139,165] : 1
x   CAL_MAG_SIDES [140,166] : 63
```

 - CAL_MAG0_ROT [80,106] : -1 (internal)
 - CAL_MAG1_ROT [95,121] : 6
 - CAL_MAG2_ROT [110,136] : 0
 - CAL_MAG3_ROT [125,151] : 6

DonLakeFlyer, 2020-08-19:
Support for 4 mags must be new. QGC only supports 3.

dagar, 2020-08-19:
> Support for 4 mags must be new. QGC only supports 3.

No, it was added years ago for the Pixhawk 2 Cube because it has 3 internal mags.

DonLakeFlyer, 2020-08-19:
It's all "new" to me if nobody tells me.

dagar, 2020-08-20:
> It's all "new" to me if nobody tells me.

Ok, let's find a better way to sync on these things going forward.

For the mags we've gotten away with it so far because the external mags were being started first, but now the internal mag (and all sensors on the board) are started before optional externals are probed.

dagar, 2020-09-04:
Thanks Don.

closed 2020-09-04 by dagar