mavlink/qgroundcontrol #9023: FORCED TO RESET ERROR

beyzaarisan, 2020-08-24:
[image]
We had connected Rapberry Pi 3 and Pixhawk to QGroundControl but QgroundControl have an error like "WARNING: EKF2 IMU1 Forced Reset". We have no time, I hope you're going to answer as soon as possible...
Regards,
Beyza

DonLakeFlyer, 2020-08-24:
These messages come from the Vehicle. This is a firmware problem

closed 2020-08-24 by DonLakeFlyer

beyzaarisan, 2020-08-24:
> These messages come from the Vehicle. This is a firmware problem

Should we buy a new one or something? Or just is it enough to re-update Pixhawk's firmware from QGroundControl?

hamishwillee, 2020-08-24:
@beyzaarisan What Don is saying is that you're asking in the wrong place. This message comes from the flight stack and just gets displayed by QGC. 

What you need to do is go ask in the support page for the flight stack you are using - presumably PX4: https://discuss.px4.io/
Remember to post what pixhawk you are using and what version of PX4 you are using.

Just updating the firmware from QGC will probably not work.

beyzaarisan, 2020-08-25:
> @beyzaarisan What Don is saying is that you're asking in the wrong place. This message comes from the flight stack and just gets displayed by QGC.
>
> What you need to do is go ask in the support page [...]

OK. Got it. Thanks for your help.