mavlink/qgroundcontrol #9028: vs2017+qt5.12.6

mengchenyi, 2020-08-24:
I used vs2017+qt5.12.6 to compile the code, but the following error occurred：**_Error while building/deploying project qgroundcontrol (kit: Desktop Qt 5.12.6 MSVC2017 64bit)
The kit Desktop Qt 5.12.6 MSVC2017 64bit has configuration issues which might be the root cause for this problem.
When executing step "Make"_**

I've been using QT for more than 4 years, but this problem has plagued me all day，who can help me，I will be very grateful！

mengchenyi, 2020-08-24:
My OS is windows10. I configured the environment exactly according to the official recommendations like this:https://dev.qgroundcontrol.com/master/en/getting_started/index.html

DonLakeFlyer, 2020-08-24:
As is states in the Issue text. This is not the place for build problems. Best place to get access other QGC devs is the QGC slack channel

closed 2020-08-24 by DonLakeFlyer