mavlink/qgroundcontrol #9060: Cannot connect multiple vehicles over serial communication

dhruv-mehta99, 2020-09-08:
The following template is for Issue/Bug reporting only. https://docs.qgroundcontrol.com/en/Support/Support.html#github-issues

## Current Behavior
the error is when I try to connect more than one vehicle to QGC then the software only reads data over serial communication from 1 vehicle only (the vehicle which is connected 1st) and doesn't read data from 2nd vehicle (though vehicle connects to the QGC, but the software doesn't read data from that com port(COM3 in my case).

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. connect any  1 vehicle via serial communication.
2. now connect 2nd vehicle over serial communication.
3. the 2nd vehicle won't show up on the screen but it would make the software misbehave.

## System Information
When posting bug reports, include the following information
- Operating System: Windows 10
- QGC Version: daily build (also in stable v 4.0)
- QGC build: this issue is in all builds of all versions
- Flight Controller: Pixhawk 2.4.8

## Detailed Description

the readBytes function in SerialLink.cc file is not called to read data from COM3 ( as a result of which 2nd vehicle heartbeat is not received by the multivehicle manager and so the multivehicle mode UI is not activated in frontend).
as a result of this whole situation, the software misbehaves when more then 1 vehicle is connected via serial communication.

Though when I connect these vehicles individually on the same ports(COM4 and COM3) they work fine.

DonLakeFlyer, 2020-09-14:
The code for de-duping composite USB devices is screwed up.

DonLakeFlyer, 2020-09-15:
@dhruv-mehta99 There should be new daily builds in about an hour. Can you verify the fix?

closed 2026-10-06 by github-actions[bot]