mavlink/qgroundcontrol #9068: improve documentation about ADS-B

booo, 2020-09-14:
Documentation about ADS-B setup is very minimal:

https://docs.qgroundcontrol.com/master/en/SettingsView/General.html#adbs_server

We should improve the documentation.

If I get some pointers in the right direction I can setup a test system and write some documentation.

DonLakeFlyer, 2020-09-14:
@antiheavy Wondering if you guys ever wrote up any docs for this for you Sentera usage?

It's similar to this: https://discuss.ardupilot.org/t/ads-b-mission-planner-and-flightaware-ads-b-dongle/21362

Basically an SDR, and then dump1090 spitting out "sbs" format if I remember correctly. Some more details in the pull: https://github.com/mavlink/qgroundcontrol/pull/8046

hamishwillee, 2020-09-15:
FYI, I just fixed up the spelling of that section, which was all "ADBS" not "ADSB" - when that fix propagates the docs pointed to above will move link to https://docs.qgroundcontrol.com/master/en/SettingsView/General.html#adsb_server

Yes it is minimal. I have no idea how this works. Would be great if we can get some information on how it should work so that this can be documented. Thanks @booo !

booo, 2020-09-15:
I used https://github.com/antirez/dump1090 with the --net option. Worked fine. I saw the plane in QGC.

booo, 2021-09-01:
I think the issues is fixed with https://github.com/mavlink/qgc-user-guide/pull/277

closed 2021-09-01 by booo