mavlink/qgroundcontrol #9075: Fullscreen on Mac Version like 3.xx

taileron, 2020-09-16:
V4.0.x now offers fullscreen mode for Android which is nice but with additional tools 3.x offered it as well

On Mac OS V 3.x has the most perfect fullscreen support - that´s one reason I use Mac OS.

V4.0.x doesn´t offer fullscreen on Mac OS anymore not even with additional tools or workarounds, is there a plan to reintegrate it?

I also wanted to show my enthusiasm about the fact that the Daily Build now runs simultaneously on the same Mac system without automatically deleting the respective settings again and again.

DonLakeFlyer, 2020-09-16:
If you are talking about full screen for video: Fullscreen video is available again on daily builds by double-clicking the video feed when it is non in PIP mode.

taileron, 2020-09-16:
i mean the whole window of QGC now always has its own top bar and always leaves the title bar and dock and so on, so that at 1280x800 there is only little space left for the content

DonLakeFlyer, 2020-09-16:
I don't understand it you are talking about QGC ui or OS ui. If you are using a daily build and are in full video. Double-clicking the video will cause all the QGC ui widgets to go away.

taileron, 2020-09-17:
In Mac OS clicking on the green one of the three upper left buttons the application changes from window to fullscreen if you hold shift the window only fits to the space that´s available (remains windowed) and QGC 4.x behaves like the shift is always hold and denies the real fullscreen mode.

dogmaphobic, 2020-09-17:
I haven't tested this in a while. Maybe it's the "double click the toolbar" to make it borderless fullscreen?

taileron, 2020-09-17:
this function was definitely lost somewhere since 3.5
Most of the other issues such as hiding elements from the video etc. have been solved for a while.
In MacOs, double-clicking on the title bar is by default set to Place in Dock.
Double-click for FS is only available in Windows.

DonLakeFlyer, 2020-09-17:
Ok, understand now.

taileron, 2020-09-17:
Fullscreen support was suddenly available on the Android platform above 4.0.x and on Mac it was gone. Has the version of QT between 3.5.x and 4.0.x been especially changed?
As already mentioned, it's great that daily 4.0 can now be used at the same time as the 3.x release, so it's not too bad at the moment.

closed 2026-10-06 by github-actions[bot]