mavlink/qgroundcontrol #9078: Disable mission failure message boxes

tzuriel-spearuav, 2020-09-17:
Working with the QGroundControl code, I get into situations where a mission failure message box appears, for example "Unable to start mission: Vehicle failed to arm."
This error often comes multiple times very fast, which effectively doesnt allow me to close them.
I dug into the code and found that its implemented not via a modal message box mechanism, but as a qml MessageDialog element. 

**Suggestion**: have a setting flag to bypass those message boxes and direct the message into the vehicle message log. 
I see similar bypass is already implemented for unit tests (`QGCApplication::showAppMessage`)

DonLakeFlyer, 2020-09-21:
> have a setting flag to bypass those message boxes and direct the message into the vehicle message log.

That is not the right way to deal with this. Vehicle messages and applications messages must be kept separate. Otherwise the user has no idea where they came from. The correct fix is to not display it multiple times. If you can detail steps to reproduce I can stop it from showing up multiple times.

tzuriel-spearuav, 2020-09-22:
The messages I've got are "NAV_CMD_COMPONENT_ARM_DISARM command failed" and "Unable to start mission: Vehicle failed to arm".
Don't have repro scenario.

DonLakeFlyer, 2020-09-22:
What firmware/version/vehicle type?

tzuriel-spearuav, 2020-09-23:
I'm using SITL firmware to simulate a quad

closed 2020-11-22 by DonLakeFlyer

julianlopcor, 2021-09-28:
> **Suggestion**: have a setting flag to bypass those message boxes and direct the message into the vehicle message log. I see similar bypass is already implemented for unit tests (`QGCApplication::showAppMessage`)

I'm trying to disable or hide the error messages, how did you do it exactly? Thanks in advance!

tzuriel-spearuav, 2021-10-06:
@julianlopcor the `QGCApplication::showAppMessage` function bypasses these message boxes when app is ran for unit testing scenario. You can hack the code to do the same and spill out these messages into log instead of invoking  `showMessageDialog`