mavlink/qgroundcontrol #9079: Wrong battery status

cuhome, 2020-09-18:
I am testing the qgc built daily and found several bugs in the battery status:
First: When the battery is not connected, the charge state displays OK, which will mislead the user that the battery connection has been completed.
Second: qgc cannot display uavcan battery voltage, because UAVCAN battery message does not include battery_cell.
qgc version：
[image]

[image]

DonLakeFlyer, 2020-09-18:
> qgc cannot display uavcan battery voltage, because UAVCAN battery message does not include battery_cell.

I don't know what this means? What exactly goes wrong?

DonLakeFlyer, 2020-09-18:
> qgc cannot display uavcan battery voltage, because UAVCAN battery message does not include battery_cell.

If you mean the mavlink 'BATTERY_STATUS' message in this case does not include voltage data. Then isn't that a firmware bug?

cuhome, 2020-09-19:
[image]
The old version of the battery data reads this information, the new version reads this and the following information.
[image]

[image]

cuhome, 2020-09-19:
If I modify the firmware of the flight controller, can I modify it like this?

DonLakeFlyer, 2020-09-19:
You'd have to ask the firmware folks about that. Would be good the enter an issue against firmware on this for discussion.

DonLakeFlyer, 2020-09-19:
> When the battery is not connected, the charge state displays OK, which will mislead the user that the battery connection has been completed.

How did you get in to this state. How can the battery not be connected yet the vehicle is still connected?

cuhome, 2020-09-21:
I use usb to connect the flight controller to the computer. Another problem I described is the problem of the qgc and px4 firmware coordination. For the 1.11.0 firmware, run the daily build qgc and use uavcan battery, whatever This failure occurs on all boards. If it is repaired by modifying the firmware, the user will need to wait for half a year. Unfortunately, the CUAV X7 series flight controller cannot use the 4.0.10 qgc; so it must use the daily built qgc, and it uses uavcan batteries by default.

DonLakeFlyer, 2020-09-21:
LIke I said this needs to move to a firmware issue and discussed there. QGC is doing things correctly with respect to mavlink spec as far as I can tell. Whereas firmware seems to have a significant bug in it. It's doesn't take 6 months to release point releases.

DonLakeFlyer, 2020-09-21:
I'm closing the issue on the QGC side until someone can show that QGC is doing something wrong.

closed 2020-09-21 by DonLakeFlyer