mavlink/qgroundcontrol #9080: Language setting on Daily Builds

farid002, 2020-09-18:
Language remains same (system - english) although I can am trying to switch it to Azerbaijani or Turkish (or any other language). Azerbaijani language is fully translated by me and my friend on Crowdin a few weeks ago.
-OS = Windows 10
-QGC= Daily Builds
[image]
[image]

farid002, 2020-09-23:
Additionally, we have checked on Android, language sources for Turkish and Azerbaijani can not be installed. Here are some screenshots @DonLakeFlyer 
[image]
[image]

DonLakeFlyer, 2020-10-08:
[image]

Sorry for taking so long. Thanks for the translations.

closed 2020-10-08 by DonLakeFlyer

farid002, 2020-10-14:
Thank you @DonLakeFlyer !