mavlink/qgroundcontrol #9083: Qt version and MacOS compatibility

tmcsys, 2020-09-20:
Is there any v4.0.x version pf QGC that will run oln MacOS 10.11.6?

The QGC documentation indicates that the latest version should install on any Mac running 10.10 or later, but v4.0.10 does not run on MacOS 10.11.6.  After a seemingly endless and definitely unsuccessful startup, I tried starting the app from the command line and got this message:

Sorry, "QGroundControl" can not be run on this version of macOS. Qt requires macOS 10.12.0 or later, you have MacOS 10.11.6.

I was able to successfully install v3.5.6 successfully. (I realize my OS version is getting a little old in the tooth.)

There is not a lot of information about System requirements n the Release Notes.

At any rate, the documentation should probably be updated.

DonLakeFlyer, 2020-09-20:
It's pretty much whatever is required by Qt. So only older builds will work on OS older than 10.12. I'll update the docs.

tmcsys, 2020-09-20:
Do you know when the Qt version changed?  

Was it v4.0.0 so v3.5.6 is as current as I'm going to get?   Or did it change sometime after that (I'm hoping)?

Thanks.

DonLakeFlyer, 2020-09-21:
Qt changed with 4.0. There is nothing newer than 3.5.6 with the old Qt.

DonLakeFlyer, 2020-09-21:
Documentation has been updated

closed 2020-09-21 by DonLakeFlyer