mavlink/qgroundcontrol #9085: Restore previous map location on start

akontsevich, 2020-09-23:
## Expected Behavior
On QGC start show map on location saved on exit. 

## Current Behavior
On start always shows me some location in Kiev for some reason.

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. Set map to some desired location
2. Exit QGC
3. Restart QGC
4. Location on the map is different from previous

## System Information
When posting bug reports, include the following information
- Operating System: Ubuntu 20.04
- QGC Version: 9.3
- QGC build: daily

## Detailed Description
Previously recent location was restored correctly on start.

akontsevich, 2020-09-23:
This problem is the caused by this code: [FlightMap.qml#L86-L87](https://github.com/mavlink/qgroundcontrol/blob/master/src/FlightMap/FlightMap.qml#L86-L87) Which I think should be changed to

```
if(QGroundControl.settingsManager.flyViewSettings.keepMapCenteredOnVehicle.rawValue ||
   _activeVehicleCoordinate.isValid)
    center = gcsPosition

```
i.e. remove negation **!** before `_activeVehicleCoordinate.isValid`.

akontsevich, 2020-09-23:
Also GCS reports its position to QGC so QGC centers map to `gcsPosition` coordinate value which shown on map as `QGCLogoFull.svg` if `gcsHeading` value is `NaN` and `QGCLogoArrow.svg` - otherwise: [FlightMap.qml#L129](https://github.com/mavlink/qgroundcontrol/blob/master/src/FlightMap/FlightMap.qml#L129). On desktop GCS position reported wrong (ISP provider location or something) so I think it is useless there.

DonLakeFlyer, 2020-09-24:
Can your verify what happens for you with the same repro steps on the Stable build?

akontsevich, 2020-09-24:
Same bug on stable.

akontsevich, 2020-09-24:
So https://github.com/mavlink/qgroundcontrol/pull/9086 fix is not correct for You, @DonLakeFlyer?

DonLakeFlyer, 2020-10-06:
QGC is working as expected: If there is a gcs location and no vehicle location the map will center to the gcs location.

closed 2020-10-06 by DonLakeFlyer

akontsevich, 2020-10-08:
> QGC is working as expected: If there is a gcs location and no vehicle location the map will center to the gcs location.

And if there is no gcs location and no vehicle, @DonLakeFlyer?! This happens on a desktop and simulator environment or if GCS does not report its position and just need to restore previous location. It should restore map to previous location like it worked some time ago. Now it is not the case as GCS location is not correct. Please reopen.

DonLakeFlyer, 2020-10-08:
> And if there is no gcs location and no vehicle

In that case QGC should put the map in the last position it was left at.

akontsevich, 2020-10-08:
> > And if there is no gcs location and no vehicle
> 
> In that case QGC should put the map in the last position it was left at.

Bu it does not do that! That is what this bug about. Please reopen.

DonLakeFlyer, 2020-10-08:
> In that case QGC should put the map in the last position it was left at.

I have verified on both Stable and Daily that this is working correctly. If you have a "Q" icon on you map then it means your OS is reporting a device location. QGC uses this as GCS location. As I have said before if there is a GCS location it will move to that. If you don't want that then you need to turn off location support in your OS somehow. For example on OSX you can test this easily but just turning off WiFi.

DonLakeFlyer, 2020-10-08:
The reason for this feature is that the creation of Plans happens quite often at the location you plan to fly from. Hence you are positioned at that location.

akontsevich, 2020-10-08:
>The reason for this feature is that the creation of Plans happens quite often at the location you plan to fly from. Hence you are positioned at that location.

I know, and that is why I need it, however feature is broken.

> I have verified on both Stable and Daily that this is working correctly. If you have a "Q" icon on you map then it means your OS is reporting a device location. 

My Ubuntu Laptop does not have any GPS or any other location info. So Q icon appears in 600 km from my place which is incorrect and QGC does overrides saved map location.

DonLakeFlyer, 2020-10-08:
The fact that your device/os reports a crap device/os location is the devices problem not QGCs problem. If you don't want it to do that then turn off location services in your OS.

akontsevich, 2020-10-08:
No, I have removed `geoclue` and QGC prints warning to console:

```
qt.positioning.geoclue2: Unable to obtain the client patch: "org.freedesktop.DBus.Error.ServiceUnknownThe name org.freedesktop.GeoClue2 was not provided by any .service files"
QGCPositionManager error 0
```

However still does not restore previous map location correctly.

DonLakeFlyer, 2020-10-08:
This is because you are building QGC yourself instead of using a pre-built AppImage. If you build it yourself and run from Qt Creator QGC currently does not provide a .desktop file in the right location which in turn will make position services work correctly. It's a known problem which I haven't had time to fix yet since it isn't critical. I have it on my list.

akontsevich, 2020-10-08:
DO You mean: "make position services work **in**correctly"?! It does not work for me if I run QGC from QtC. How desktop file may affect this issue?! And it worked correctly 6 month ago until this GCS icon appeared.

akontsevich, 2020-10-08:
Also QGC does not have an option to hide this Q icon completely if no location provided.

akontsevich, 2021-02-12:
> This is because you are building QGC yourself instead of using a pre-built AppImage. 

No AppImage also shows me wrong arbitrary location - 600 km from my place, @DonLakeFlyer.