mavlink/qgroundcontrol #9100: Toolbar buttons - when selecting a button in touch screen (tablet) the highlight effect stays on button

tzuriel-spearuav, 2020-10-05:
## Expected Behavior
Highlight effect on button should disappear upon clicking it

## Current Behavior
When hovering a button on image toolbar (Fly, Return, etc) - on windows with mouse pointing device the button shows hover effect highlighting the button with color, this highlight vanish when mouse cursor is no longer on the button. In touch screen devices, as as tablet, pressing the button "Return", which causes confirmation slider to appear, the button stays highlighted until user touch any other toolbar button on screen.

## Steps to Reproduce:
1. Install latest nightly build of QGC on a tablet computer (I'm using a windows tablet)
2. Connect copter or SITL simulator
3. During mission, press with thumb on "Return" button (RTL)
4. Notice return is highlighted while confirmation slider is shown and sometimes even after dismissing the slider (confirming or cancelling)

## System Information
- Win10
- QGC nightly
- Flight Controller: Ardupilot SITL (copter)

tzuriel-spearuav, 2020-10-06:
Comment: disabling the hovered effect from highlight mechanism prevents this problem (in the expense of removing the hover effect of course)
```
QGCButton {
    ...
    _showHighlight:     pressed | /*hovered |*/ checked
    ...
}
```

DonLakeFlyer, 2020-10-28:
Does this only happen on the ToolStrip buttons on the left? What about normal buttons like "Ok" buttons in dialog?

tzuriel-spearuav, 2020-10-29:
With OK/Cancel buttons, the problem cannot be seen because a dialog would then be dismissed. I'll have to re-test this on a tablet with other toolbars. However, I think QGCHoverButton is only used in this left toolbar

DonLakeFlyer, 2020-10-29:
Duh, yeah, that was stupid. Needs to be a place which does not dismiss. Here is one:
* Application Settings
* Scroll to the bottom
* In Brand Image try clicking "Reset Default Branding Image"

DonLakeFlyer, 2020-10-29:
I"m trying to determine if this is a Qt bug or a QGC bug.

tzuriel-spearuav, 2020-11-01:
Just check with "Reset Default Branding Image" and indeed every other click on the button - the button stays with highlight color until you click anywhere else - on background or any other button

DonLakeFlyer, 2020-11-02:
Hmm, that's not good news. Points more to a Qt bug. Hopefully there is some way I can work around.