mavlink/qgroundcontrol #9106: Qgroundcontrol on iOS

taileron, 2020-10-09:
I like to use QGC on iOS and currently all features I need are only offered in V 3.4.3
In a iOS 10.3.3 (iPhone 6s) unit that´s running absolutely perfect
Now I managed with extreme effort to get another still on iOS 10 unit ... but now only V3.2 is available from the Appstore, the only version that works properly for flight monitoring is 3.4.x how do I get an IPA File of this version?  
Is there a possibility to offer V3.4.3 in the iOS Appstore instead of 3.2 as a previous version below 4.0.x ?
Updated iOS units now only install 4.0.x

DonLakeFlyer, 2020-10-09:
Only the latest stable is available from the store. That isn't going to change.

Why do you need to use an older build?

closed 2020-10-09 by DonLakeFlyer

taileron, 2020-10-09:
... the only version with such low rtsp quality/latency that allows 16:9 full screen fpv flying with easy hiding hud and taskbar by touch the background.

Purchased lot´s of hardware stuff like analog to rtsp encoder and multi antenna receivers to get all that working after I noticed how perfect the rtsp client of V3.4.x was. e.g. 3D printed parts like Phone holder to the RC TX and so on ...
The Appstore still offers 3.2 for old iOS devices but no 3.4

(on Mac I use QGC daily for configuring and 3.4.4 for flight monitoring / rtsp video but the Mac does not fit into my FPV optics...) 

And in many respects the latest iOS 4.0.X is still far from being able to be used for fpv

iOS 4.0.10
[image]
iOS 3.2
[image]