mavlink/qgroundcontrol #9114: Custom example android build fails

jlecoeur, 2020-10-15:
Steps to reproduce
```
git clone https://github.com/mavlink/qgroundcontrol
cd qgroundcontrol
git submodule update --init
docker run -it --rm -w $PWD --env=LOCAL_USER_ID="$(id -u)" --volume=$PWD:$PWD:rw mavlink/qgc-build-android:latest /bin/bash
cp -r custom-example custom
cd custom
python2 updateqrc.py
cd ../
mkdir -p build/android
cd build/android
qmake ../../qgroundcontrol.pro CONFIG+=installer -spec android-clang
make -j 8
```

Output:
```
jlecoeur@FLPC005:~$ git clone https://github.com/mavlink/qgroundcontrol
Cloning into 'qgroundcontrol'...
remote: Enumerating objects: 4, done.
remote: Counting objects: 100% (4/4), done.
remote: Compressing objects: 100% (4/4), done.
remote: Total 140871 (delta 0), reused 0 (delta 0), pack-reused 140867
Receiving objects: 100% (140871/140871), 378.88 MiB | 1.75 MiB/s, done.
Resolving deltas: 100% (107330/107330), done.
jlecoeur@FLPC005:~$ cd qgroundcontrol
jlecoeur@FLPC005:~/qgroundcontrol$ git submodule update --init
Submodule 'libs/OpenSSL/android_openssl' (https://github.com/Auterion/android_openssl) registered for path 'libs/OpenSSL/android_openssl'
Submodule 'libs/mavlink/include/mavlink/v2.0' (https://github.com/mavlink/c_library_v2.git) registered for path 'libs/mavlink/include/mavlink/v2.0'
Submodule 'libs/qmlglsink/gst-plugins-good' (https://github.com/mavlink/gst-plugins-good.git) registered for path 'libs/qmlglsink/gst-plugins-good'
Submodule 'src/GPS/Drivers' (https://github.com/PX4/GpsDrivers.git) registered for path 'src/GPS/Drivers'
Cloning into '/home/jlecoeur/qgroundcontrol/libs/OpenSSL/android_openssl'...
Cloning into '/home/jlecoeur/qgroundcontrol/libs/mavlink/include/mavlink/v2.0'...
Cloning into '/home/jlecoeur/qgroundcontrol/libs/qmlglsink/gst-plugins-good'...
Cloning into '/home/jlecoeur/qgroundcontrol/src/GPS/Drivers'...
Submodule path 'libs/OpenSSL/android_openssl': checked out '3aaff1bd9e35047abdb363239bb3e3c114d07ea1'
Submodule path 'libs/mavlink/include/mavlink/v2.0': checked out 'd1eea85b685eec2204ad31c7f7f51486f218f86b'
Submodule path 'libs/qmlglsink/gst-plugins-good': checked out '9d782fad9dc0384ba86ecae64511c193f6149f93'
Submodule path 'src/GPS/Drivers': checked out '318562bac7cd55e6cfafd4694094343a70e4f322'
jlecoeur@FLPC005:~/qgroundcontrol$ docker run -it --rm -w $PWD --env=LOCAL_USER_ID="$(id -u)" --volume=$PWD:$PWD:rw mavlink/qgc-build-android:latest /bin/bash
cp -r custom-example custom
cd custom
python2 updateqrc.py
cd ../
mkdir -p build/android
cd build/android
qmake ../../qgroundcontrol.pro CONFIG+=installer -spec android-clang
make -j 8Starting Xvfb
Starting with UID : 1000
user@8a04b2c122ab:/home/jlecoeur/qgroundcontrol$ cp -r custom-example custom
user@8a04b2c122ab:/home/jlecoeur/qgroundcontrol$ cd custom
user@8a04b2c122ab:/home/jlecoeur/qgroundcontrol/custom$ python2 updateqrc.py
bash: python2: command not found
user@8a04b2c122ab:/home/jlecoeur/qgroundcontrol/custom$ cd ../
user@8a04b2c122ab:/home/jlecoeur/qgroundcontrol$ mkdir -p build/android
user@8a04b2c122ab:/home/jlecoeur/qgroundcontrol$ cd build/android
user@8a04b2c122ab:/home/jlecoeur/qgroundcontrol/build/android$ qmake ../../qgroundcontrol.pro CONFIG+=installer -spec android-clang
Info: creating stash file /home/jlecoeur/qgroundcontrol/build/android/.qmake.stash
Project MESSAGE: Qt version 5.12.6
Project MESSAGE: Android Arm 32 bit build
/usr/bin/ccache
Project MESSAGE: Found ccache, enabling
Project MESSAGE: Daily Build
Project MESSAGE: QGroundControl Development master:e5f4bfe 2020-10-15 10:07:59 -0700
Project MESSAGE: Build Installer
Project MESSAGE: Release flavor
Project MESSAGE: Found custom build
Project MESSAGE: Adding Custom Plugin
Project MESSAGE: Custom QGC Version: 0.0.5951
[... 1269 lines cut ...]
compiling src_FactSystem_FactControls_FactBitmask_qml.cpp
compiling src_FactSystem_FactControls_FactLabel_qml.cpp
compiling src_FactSystem_FactControls_FactTextFieldRow_qml.cpp
compiling src_FactSystem_FactControls_FactValueSlider_qml.cpp
compiling src_FactSystem_FactControls_FactCheckBox_qml.cpp
compiling src_FactSystem_FactControls_FactTextField_qml.cpp
compiling src_FactSystem_FactControls_FactTextFieldSlider_qml.cpp
compiling src_FactSystem_FactControls_AltitudeFactTextField_qml.cpp
compiling src_FactSystem_FactControls_FactTextFieldGrid_qml.cpp
compiling src_FactSystem_FactSystemTest_qml.cpp
compiling src_FirstRunPromptDialogs_UnitsFirstRunPrompt_qml.cpp
compiling src_FirstRunPromptDialogs_OfflineVehicleFirstRunPrompt_qml.cpp
compiling src_FlightDisplay_SubChecklist_qml.cpp
compiling src_FlightDisplay_RoverChecklist_qml.cpp
compiling src_FlightDisplay_VTOLChecklist_qml.cpp
compiling src_FlightDisplay_MultiRotorChecklist_qml.cpp
compiling src_FlightDisplay_FixedWingChecklist_qml.cpp
compiling src_FlightDisplay_DefaultChecklist_qml.cpp
compiling src_ui_toolbar_JoystickIndicator_qml.cpp
compiling src_ui_toolbar_GPSIndicator_qml.cpp
compiling src_ui_toolbar_TelemetryRSSIIndicator_qml.cpp
compiling src_ui_toolbar_MainToolBarIndicators_qml.cpp
compiling src_ui_toolbar_LinkIndicator_qml.cpp
compiling src_ui_toolbar_ModeIndicator_qml.cpp
compiling src_

DonLakeFlyer, 2020-10-15:
That's certainly interesting :)

DonLakeFlyer, 2020-10-16:
This is working fine for me building with Qt Creator. It seems like you are putting the build dir inside the repo. Are you sure that isn't the problem? Also does config+=installer have anything to do with it?

jlecoeur, 2020-10-17:
Good news that it works with Qt Creator. I would really like to make it work in CI though.

- building outside the repo leads to the same problem
- if I remove CONFIG+=installer. Then the build completes, but no apk is generated.

What is strange is that the problem occurs only with the custom example.

DonLakeFlyer, 2020-10-19:
> Good news that it works with Qt Creator. I would really like to make it work in CI though.

There is no major magic in Qt Creator. If you can get Qt Creator to work for you then it should be able to tell you the qmake command line you can use instead. It's under Projects.

I'll try it again.

DonLakeFlyer, 2020-10-20:
So the main problem is that there is no support for creating custom build play store packages, which is what CONFIG+=installer does. The build scripts only work for QGC CI play store uploads. It's possible to fix this, but not anything I'll get to soon.

I hacked the installer build scripts just to see how far I could get with a Qt Creator build to work around this. My build directory is outside the repo. It all worked fine and got all the way to androiddeployqt without problems. If you do the same and get it working with Qt Creator with modified build scripts for store packaging you should be able to use the qmake command line which Qt Creator tells you about in "Effective qmake call":

[image]

DonLakeFlyer, 2020-11-20:
Closing since at this point this is simply not supported

closed 2020-11-20 by DonLakeFlyer