mavlink/qgroundcontrol #9125: Solo video is not started(anymore) - worked before

AndKe, 2020-10-20:
## Expected Behavior
Video from 3DR Solo and OpenSolo4 works

## Current Behavior
Video does not start, unless user do a 

## Steps to Reproduce:
1. Connect to a OpenSolo4.0 with video setting UDP h264, port 5600
2.observe missing video
3.issue  "nc 10.1.1.1 5502" in command line
4.video appear

## System Information
- Operating System: [Ubuntu 20.04]
- QGC Version: [today, HEAD]
- QGC build: [HEAD]
- Flight Controller: [Cube]
- Autopilot (with version): AC4.0.3]

## Detailed Description
The fix wil most likely to be highly related to https://github.com/mavlink/qgroundcontrol/issues/5870

## Log Files and Screenshots
Please see attached log
[qgclog.txt](https://github.com/mavlink/qgroundcontrol/files/5406895/qgclog.txt)

DonLakeFlyer, 2020-10-29:
Hmm, I wonder if it's time to stop trying to do this automagically and just have a setting for Solo Video in the video dropdown. For Solo it's always going to be UDP h264, port 5600 right? What do you think?

DonLakeFlyer, 2020-11-20:
@AndKe I believe the solo video start problem is fixed now

DonLakeFlyer, 2020-11-20:
Can you enter a separate issue for the crash? With a stack trace if you have one

closed 2020-11-20 by DonLakeFlyer

AndKe, 2020-11-20:
should I have some extra compilation parameters, or gdb settings when reproducing ?

DonLakeFlyer, 2020-11-21:
Not that I know of, but Linux is pretty foreign to me. Could you also verify the prebuilt released version of daily has the same crash. Want to make sure it's not just the version you built.

AndKe, 2020-11-21:
I tried to reproduce using the current master compiled here, before the appimage.
With a real 3D Solo, I was unable to make it crash when creating an nanual connetion.
I see one deficiency that should be fixed: Selecting Solo as video source does nothing until QGC is restarted. 
Once QGC is restarted - it works.
Please make Solo as video source work instantly, for consistency with other streaming modes that do not require restart to work.

Later, I will test with non-solo Arducopter and the IMX6 and let you know..

AndKe, 2020-11-21:
Ok, I can no longer reproduce the crash on my compiled version, or the appimage, nor with ArduCopter-Solo controller or Arduplane + IMX6 (not that it should matter anymore.)
So.. what remains is to apply Solo connection instantly when Solo is selected.

DonLakeFlyer, 2020-11-21:
> I can no longer reproduce the crash on my compiled version, or the appimage

Hmm. Not sure that is specifically a good thing :)

> Selecting Solo as video source does nothing until QGC is restarted.
> Once QGC is restarted - it works.

That is correct. That is the way it currently works unfortunately. Removing the need for that is tricky. So this was a quick fix to get something working at least. I'll add an Issue for 4.2 to try to fix this.

DonLakeFlyer, 2020-11-21:
https://github.com/mavlink/qgroundcontrol/issues/9195

AndKe, 2020-11-21:
Thank you- in the meantime, maybe you could change "Solo" choice to to "Solo.(req.restart)"  or something like that.

DonLakeFlyer, 2020-11-21:
Ah, I was originally trying to figure out a simple way to let the user know since it's kind of crap. Of course I didn't stumble myself onto something as simple as that!