mavlink/qgroundcontrol #9126: Mavlink inspector removes graphs when moved to free-floating window

AndKe, 2020-10-20:
## Expected Behavior
Selected graphs should be retained

## Current Behavior
All selections are cleared 

## Steps to Reproduce:
1. Open Amvlink inspector
2.graph at least one value
3.click the upper right corner to detach window
4.observe that the graph, selected values are gone.

## System Information
Operating System: [Ubuntu 20.04]
QGC Version: [today, HEAD]
QGC build: [HEAD]
Flight Controller: [Cube]
Autopilot (with version): AC4.0.3]

DonLakeFlyer, 2020-10-29:
Interesting, but complicated to fix.

AndKe, 2020-10-29:
If hard to fix, did you give the thought about free-floating by default a thought ? that would be a workaround.

Have you considered adding the graphing features of APMplanner2 ? - those are beautifully powerful. :)  
(the realtime is not as advanced, but the log analysis graphing is the best.)

DonLakeFlyer, 2020-10-29:
> Have you considered adding the graphing features of APMplanner2 ? - those are beautifully powerful. :)

Not possible. Those are all QWidget based not Qml. Also source code license prevents copying/moving to QGC.